mne-tools/mne-python #10380: [DOC] Alphabetizing `docdict`

alexrockhill, 2022-02-22:
I think the original organization of `docdict` was great when there were limited entries but now, in my opinion, it's getting a bit unruly and disorganized. My suggestion would be to alphabetize and document the standardized format that the argument should go first with any specific separated by and underscore, i.e. there can be `trans` and `trans_not_none` which will be next to each other alphabetically and that way it will be easy to tell which docstring (i.e. (`%(color_matplotlib)`) goes to which argument (i.e. `color`) where `%(matplotlib_color)` for the argument color is a bit confusing that it goes to the argument `color` for  a function such as `f(..., color=None, ...)`.

cc @hoechenberger @agramfort

agramfort, 2022-02-22:
sounds like a good idea !

> Message ID: ***@***.***>
>

hoechenberger, 2022-02-22:
+1 if you volunteer to do it 😅

alexrockhill, 2022-02-22:
Hey @andylikescodes, I make this issue so I'm happy to do a PR, but I thought it might be a good one for you to do your first pull request. If you're interested, here are the steps:

Make sure you have a fork as origin and the mne main repository as upstream with
```
git remote -v
```

It should look like this:
```
origin	https://github.com/alexrockhill/mne-python.git (fetch)
origin	https://github.com/alexrockhill/mne-python.git (push)
upstream	https://github.com/mne-tools/mne-python.git (fetch)
upstream	https://github.com/mne-tools/mne-python.git (push)
```

Switch to your main branch if you aren't on it already:

```
git checkout main
```

Get any new changes to the main branch:
```
git pull upstream main
```

Make a new branch called something like `docdict`
```
git checkout -b docdict
```

Copy paste around in the file `mne/utils/docs.py` so that all the `docdict` entries are in alphabetical order, maybe with

```
# %%
# A
```

section headers for each letter.

When you're done, do
```
git status
```

Make sure you only have files in red that you've changed and then do
```
git add -A
```

And commit with:
```
git commit -m "alphebetized docdict"
```

Finally:
```
git push origin docdict
```

Which should give you a link to do a pull request.

Honestly, this is definitely busywork but it would give you an introduction to the MNE-Python workflow and review process. It's nice to start with something pretty basic, otherwise you just get bogged down because there is a process of assimilating into the MNE style that takes a bit of time. If you'd rather do something else for your first PR, I wouldn't be offended in the slightest and I will go ahead and do this one, it just made me think it would be a good way to get some experience contributing for you.

alexrockhill, 2022-02-23:
Just noticed that all the topomap docdict entries are `topomap_x` which doesn't follow this convention and will be inconsistent in alphabetization, note to self to fix this

hoechenberger, 2022-02-23:
> If you're interested, here are the steps:

Oh my, no wonder contributing is so hard 😅 We should write an intro explaining how to do all of this using the GitHub GUI

alexrockhill, 2022-02-23:
Hahaha @andylikescodes said he has a good amount of previous GitHub experience so I think that might have been mostly review for him, but in general, I think that's a great idea. There is documentation about it https://mne.tools/dev/install/contributing.html but it's really not by example, it's quite long and if I were a new contributor I might not know where to start/how to do a PR.

andylikescodes, 2022-02-23:
@alexrockhill Yes, this seems like a great starting point for the journey. I would love to try it out. Thank you for all these suggestions. I will have some time later today and I will try it out. Will let you know when I am done.

andylikescodes, 2022-02-24:
@alexrockhill Do the entries in the docs.py file have some logical order? Should I just extract all the docdict entries and put them into a big section? These docdict entries are actually separated into different sections, for example, this section here:

```
# raw/epochs/evoked apply_function method
# apply_function method summary
applyfun_summary = """\
The function ``fun`` is applied to the channels defined in ``picks``.
The {} object's data is modified in-place. If the function returns a different
data type (e.g. :py:obj:`numpy.complex128`) it must be specified
using the ``dtype`` parameter, which causes the data type of **all** the data
to change (even if the function is only applied to channels in ``picks``).{}

.. note:: If ``n_jobs`` > 1, more memory is required as
          ``len(picks) * n_times`` additional time points need to
          be temporarily stored in memory.
.. note:: If the data type changes (``dtype != None``), more memory is
          required since the original and the converted data needs
          to be stored in memory.
"""
applyfun_preload = (' The object has to have the data loaded e.g. with '
                    '``preload=True`` or ``self.load_data()``.')
docdict['applyfun_summary_raw'] = \
    applyfun_summary.format('raw', applyfun_preload)
docdict['applyfun_summary_epochs'] = \
    applyfun_summary.format('epochs', applyfun_preload)
docdict['applyfun_summary_evoked'] = \
    applyfun_summary.format('evoked', '')
```

I am assuming I should not break this structure while I am reorganizing the docdict entries?

alexrockhill, 2022-02-24:
Yeah the idea would be that those are all in order so you can just copy the whole block over. Those would all be in the "A" block because the docdict key starts with an a. Okay progress, nice!

alexrockhill, 2022-02-24:
Also maybe just do a few and then make the PR so you don't do a bunch of extra work that needs to be redone.

andylikescodes, 2022-02-24:
What about this block?

```
# Verbose
docdict['verbose'] = """
verbose : bool | str | int | None
    Control verbosity of the logging output. If ``None``, use the default
    verbosity level. See the :ref:`logging documentation <tut-logging>` and
    :func:`mne.verbose` for details. Should only be passed as a keyword
    argument."""
docdict['add_frames'] = """
add_frames : int | None
    If int, enable (>=1) or disable (0) the printing of stack frame
    information using formatting. Default (None) does not change the
    formatting. This can add overhead so is meant only for debugging.
"""
```

There are "a" and "v" at the same time.

alexrockhill, 2022-02-24:
Verbose should go in v near the bottom and add frames should go in a near the top. I hope that makes sense.

andylikescodes, 2022-02-24:
Oh sorry, let me frame my question a bit more clear, what I meant was that in this blog of code:

```
# raw/epochs/evoked apply_function method
# apply_function method summary
applyfun_summary = """\
The function ``fun`` is applied to the channels defined in ``picks``.
The {} object's data is modified in-place. If the function returns a different
data type (e.g. :py:obj:`numpy.complex128`) it must be specified
using the ``dtype`` parameter, which causes the data type of **all** the data
to change (even if the function is only applied to channels in ``picks``).{}

.. note:: If ``n_jobs`` > 1, more memory is required as
          ``len(picks) * n_times`` additional time points need to
          be temporarily stored in memory.
.. note:: If the data type changes (``dtype != None``), more memory is
          required since the original and the converted data needs
          to be stored in memory.
"""
applyfun_preload = (' The object has to have the data loaded e.g. with '
                    '``preload=True`` or ``self.load_data()``.')
docdict['applyfun_summary_raw'] = \
    applyfun_summary.format('raw', applyfun_preload)
docdict['applyfun_summary_epochs'] = \
    applyfun_summary.format('epochs', applyfun_preload)
docdict['applyfun_summary_evoked'] = \
    applyfun_summary.format('evoked', '')
```

We have this block that is not docdicts:
```
# raw/epochs/evoked apply_function method
# apply_function method summary
applyfun_summary = """\
The function ``fun`` is applied to the channels defined in ``picks``.
The {} object's data is modified in-place. If the function returns a different
data type (e.g. :py:obj:`numpy.complex128`) it must be specified
using the ``dtype`` parameter, which causes the data type of **all** the data
to change (even if the function is only applied to channels in ``picks``).{}

.. note:: If ``n_jobs`` > 1, more memory is required as
          ``len(picks) * n_times`` additional time points need to
          be temporarily stored in memory.
.. note:: If the data type changes (``dtype != None``), more memory is
          required since the original and the converted data needs
          to be stored in memory.
"""
applyfun_preload = (' The object has to have the data loaded e.g. with '
                    '``preload=True`` or ``self.load_data()``.')
```

While these are docdicts:
```
docdict['applyfun_summary_raw'] = \
    applyfun_summary.format('raw', applyfun_preload)
docdict['applyfun_summary_epochs'] = \
    applyfun_summary.format('epochs', applyfun_preload)
docdict['applyfun_summary_evoked'] = \
    applyfun_summary.format('evoked', '')
```
What should I do with the first half of the code that are not docdicts?

alexrockhill, 2022-02-24:
My advice would be to keep them together

alexrockhill, 2022-02-24:
@andylikescodes, did you have a chance to open a PR? It would be a lot easier to give you advice if you just changed like one or two of the entries, opened the PR and then I'd be like "yeah, looks awesome" or "you might need to change this part because that's going to break this" and then you'd be on your way. Be sure when you do to write "Fixes x" where x is the url of this issue.

alexrockhill, 2022-03-02:
I didn't realize how big `doctdict` was already, I think this is a bit too late. What I thought might be a fun onboarding exercise  looks like the endless PR to scare away new contributors! Sorry @andylikescodes!

I wrote a script to automate it but it still took an hour of fiddling.

PR #10407 (2022-03-02): [MAINT, MRG] Alphabetize ``docdict`` to limit redundancy in the long term
Fixes #10380.
closed 2022-03-02 by larsoner