mne-tools/mne-python #10382: [BUG] Pyvista backend `quiver3d` behavior inconsistent

alexrockhill, 2022-02-22:
Here is a minimally reproducible example:

```
import mne
import os.path as op
data_path = mne.datasets.sample.data_path()  # get the sample data path
subjects_dir = op.join(data_path, 'subjects')
brain = mne.viz.Brain(subject_id='sample', hemi='both', surf='pial',
                      subjects_dir=subjects_dir, alpha=0.1)
brain._renderer.quiver3d(0, 0, 0, 0, 0, 10, 'red', 10, 'arrow')  # does nothing
brain._renderer.quiver3d(0, 0, 0, 0, 0, 10, 'red', 10, 'arrow',
                         scale_mode='scalar')  # still does nothing
brain._renderer.quiver3d(0, 0, 0, 0, 0, 10, 'red', 10, 'arrow',
                         scale_mode='scalar', scalars=[1])  # works
brain._renderer.quiver3d(0, 0, 0, 0, 0, 10, 'blue', 20, 'arrow',
                         scalars=[1])  # broken again
```

What I'm stuck on as far as fixing it is that the call goes to `_glyph` with `scalar=[1]` and the key line is 1101:

```
    if isinstance(scalars, str):
        dataset.active_scalars_name = scalars
```

Which shouldn't be a string so I have no idea how the one call that works does in fact work... Maybe @larsoner might have just a minute to take a peek.

alexrockhill, 2022-02-22:
EDIT: I missed this before

```
if scalars is not None:
     _point_data(grid)['scalars'] = np.array(scalars)
     scalars = 'scalars'
```

so it will be a string.

I'm really not sure what the intention was here so I'm going to leave this one for now.

alexrockhill, 2022-03-22:
@larsoner do you have a minute to look at this? I can do the PR but if you could just tell me what you were trying to do that would be really helpful.

larsoner, 2022-03-22:
I won't have time to look until late next week

alexrockhill, 2022-03-22:
No worries, whenever you have time, just trying to wrap up loose threads.

alexrockhill, 2022-04-04:
@larsoner, just checking on this again, should just take five minutes. I'll go ahead and write a PR to simplify it and I'll try and look and make sure there aren't any use cases that it breaks.

PR #10493 (2022-04-05): [BUG, MRG] Fix Quiver3D default behavior
Fixes #10382.
closed 2022-04-05 by agramfort