mne-tools/mne-python #10404: BUG: _compare_version backward incompat

larsoner, 2022-02-28:
cc @hoechenberger:
```
>>> import mne
>>> mne.fixes._compare_version('1.0.dev0', '>=', '1.0')
False
>>> from distutils.version import LooseVersion
>>> LooseVersion('1.0.dev0') >= LooseVersion('1.0')
<stdin>:1: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
True
```
This is a problem because we usually want the second result :(

The `git grep check_version | wc -l`  count is 67 and `git grep _compare_version | wc -l` is 26, I'd rather not have to change all of these to allow for `.dev0` extensions. Plus other submodules like `mne-nirs` that already use this (where I saw the problem actually!) will not work properly...

I'm tempted to change this to cut off any `.dev` from the end to make life easy, but then we depart from standard. But it seems a bit like the lesser of two evils here.

hoechenberger, 2022-03-01:
I believe there were some relatively recent changes by @cbrnr if I'm not mistaken (cannot check right now), could you have a look at the history of that function and see if this problem maybe only started showing up after these changes?

cbrnr, 2022-03-01:
This is a bit unfortunate and maybe even a bug? They are explicitly recommending to switch to `packaging.version.parse`, but it doesn't behave identically. I've open an issue.

hoechenberger, 2022-03-01:
I have to say that IMHO the new behavior is more correct than the old one though. 1.0 **is** newer than 1.0dev0. So the version comparison should be `>=1.0dev0`, no? I think this is how conda behaves too if I'm not mistaken.

cbrnr, 2022-03-01:
Technically yes, `dev0` is a *pre*release, so it is less than the final release. However, speaking of [semantic versioning](https://semver.org/), I think we're not using those prerelease tags correctly, because they should be separated by a hyphen (e.g. `1.0.0-dev0`).

cbrnr, 2022-03-01:
As @hoechenberger suspected this is expected behavior, and `distutils.version` was doing its own thing. I'd change the comparisons in our code. Alternatively, we could vendor the old versioning code, but I'd rather not do that.

hoechenberger, 2022-03-01:
Let's change our code to make it more consistent with my thinking 😅🥴😁

larsoner, 2022-03-01:
I'd rather add a loose=True that prunes the dev. Most of the time if a person is using 1.0.dev0, they probably have what's needed and if they don't they know how to update. Plus it's backward compatible this way. And it's much easier than changing 100+ places we use (and think about) these comparisons.

cbrnr, 2022-03-01:
So this would basically affect equality comparisons, right? Using `loose=True` would mean that `1.0.dev0 >= 1.0`, `1.0.dev0 <= 1.0`, and `1.0.dev0 == 1.0` are all `True`. Does this make sense? I don't really understand why we rely on such strange comparisons.

larsoner, 2022-03-01:
I can document this inaccuracy with `==` and `loose=True`. In practice we (almost?) never use `==`. To me practicality -- and backward compat for other libs and even users that make use of `check_version` and have silently been getting the wrong result, at least according to how this has behaved for the last 9 years -- beats purity here by a long shot.

larsoner, 2022-03-01:
The most important thing is that it restores the behavior of `check_version`. This uses `_compare_version` under the hood with a `<` operator to say if something is not okay:

https://github.com/mne-tools/mne-python/blob/6608b720b214529732542cb2aac11b401770eddf/mne/utils/check.py#L100

So for example:
```
>>> LooseVersion('1.0.dev0') < LooseVersion('1.0')
False
>>> mne.fixes._compare_version('1.0.dev0', '<', '1.0')
True
```
So I think it should work. So maybe we should add `loose=True` to `check_version`, do the stripping there, and not change `_compare_version` (and just fix the way fewer of those we use, where needed).

cbrnr, 2022-03-01:
> So maybe we should add loose=True to check_version, do the stripping there, and not change _compare_version (and just fix the way fewer of those we use, where needed).

This sounds good!

PR #10406 (2022-03-01): BUG: Fix backward compat check_version
Closes #10404

A fun little regex exercise :)
closed 2022-03-01 by agramfort