mne-tools/mne-python #10413: spectral_connectivity

mkkubins, 2022-03-04:
Hello!
I cannot access spectral_connectivity. Also when I am looking for this in the MNE search there is nothing. Is it removed now? 

[This page doesn't work for me](https://mne.tools/mne-connectivity/stable/generated/mne_connectivity.spectral_connectivity.html)

ERROR PRINTED: 

File not found

The site configured at this address does not contain the requested file.

If this is your site, make sure that the filename case matches the URL.
For root URLs (like http://example.com/) you must provide an index.html file.

[Read the full documentation](https://help.github.com/pages/) for more information about using GitHub Pages.

agramfort, 2022-03-04:
what are you looking for? the page you point to has been certainly moved

adam2392, 2022-03-08:
@mkkubins the functionality you're looking for has been moved to https://mne.tools/mne-connectivity/stable/generated/mne_connectivity.spectral_connectivity_epochs.html since v0.23. It's also undergone some changes.

If you're looking for the original implementation it's v0.1 from mne-connectivity, which is essentially a copy of the MNE api. Since then it's evolved and improved a lot. I would recommend checking https://mne.tools/mne-connectivity/dev/auto_examples/connectivity_classes.html#sphx-glr-auto-examples-connectivity-classes-py to see how the output of the connectivity functions have evolved. 

Hope that helps.

mkkubins, 2022-03-09:
Thanks for the help!

closed 2022-03-09 by mkkubins