mne-tools/mne-python #10426: Problem with brain.add_volume_labels in tutorial 'Working with sEEG data'

RuxueG, 2022-03-10:
Hi,
I met a problem when running tutorial Jupyter notebook scripts of 'Working with sEEG data'. I cannot plot out the labels in 3D head model when running the command brain.add_volume_labels(aseg='aparc+aseg', labels=labels). There's a TypeError: SetColor argument 1: expected a sequence of 3 values, got 4 values. A document for the details of the error was attached here. 
[error.pdf](https://github.com/mne-tools/mne-python/files/8219127/error.pdf)
Could you help me solve the problem? Thanks!

By the way, I worked with the 
Windows system and MNE-python in Anaconda environment.

BarryLiu97, 2022-03-10:
Please update all the packages to the latest. Use
```
mne.sys_info()
```
to see the version of your package.

RuxueG, 2022-03-10:
I tried to update all the packages but I still meet the same problem. Here is the info

Platform:       Windows-10-10.0.22000-SP0
Python:         3.9.10 | packaged by conda-forge | (main, Feb  1 2022, 21:22:07) [MSC v.1929 64 bit (AMD64)]
Executable:     F:\anaconda\anaconda\envs\mne\python.exe
CPU:            Intel64 Family 6 Model 142 Stepping 10, GenuineIntel: 8 cores
Memory:         7.9 GB

mne:            0.24.1
numpy:          1.22.3 {}
scipy:          1.8.0
matplotlib:     3.5.1 {backend=QtAgg}

sklearn:        1.0.2
numba:          0.53.1
nibabel:        3.2.2
nilearn:        0.9.0
dipy:           1.4.1
cupy:           Not found
pandas:         1.4.1
mayavi:         4.7.4
pyvista:        0.33.3 {OpenGL 4.5.0 - Build 25.20.100.6576 via Intel(R) UHD Graphics 620}
pyvistaqt:      0.7.0
ipyvtklink:     0.2.2
vtk:            9.1.0
PyQt5:          5.15.6
ipympl:         0.8.8
mne_qt_browser: 0.2.5
pooch:          v1.6.0

Is there anything that I missed?

RuxueG, 2022-03-10:
I think the latest version doesn't work based on this warning 

F:\anaconda\anaconda\envs\mne\lib\site-packages\mne\viz\backends\_pyvista.py:192: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(pyvista.__version__) >= '0.27.0':
F:\anaconda\anaconda\envs\mne\lib\site-packages\mne\viz\backends\_pyvista.py:192: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(pyvista.__version__) >= '0.27.0':
F:\anaconda\anaconda\envs\mne\lib\site-packages\mne\surface.py:1728: DeprecationWarning: Call to deprecated method ThresholdBetween. (Use 'SetLowerThreshold', 'SetUpperThreshold' and 'SetThresholdFunction' instead.) -- Deprecated since version 9.1.0.
  selector.ThresholdBetween(val, val)
F:\anaconda\anaconda\envs\mne\lib\site-packages\mne\surface.py:1728: DeprecationWarning: Call to deprecated method ThresholdBetween. (Use 'SetLowerThreshold', 'SetUpperThreshold' and 'SetThresholdFunction' instead.) -- Deprecated since version 9.1.0.
  selector.ThresholdBetween(val, val)
F:\anaconda\anaconda\envs\mne\lib\site-packages\mne\surface.py:1728: DeprecationWarning: Call to deprecated method ThresholdBetween. (Use 'SetLowerThreshold', 'SetUpperThreshold' and 'SetThresholdFunction' instead.) -- Deprecated since version 9.1.0.
  selector.ThresholdBetween(val, val)
F:\anaconda\anaconda\envs\mne\lib\site-packages\mne\surface.py:1728: DeprecationWarning: Call to deprecated method ThresholdBetween. (Use 'SetLowerThreshold', 'SetUpperThreshold' and 'SetThresholdFunction' instead.) -- Deprecated since version 9.1.0.
  selector.ThresholdBetween(val, val)

But how could I deal with the problem?

BarryLiu97, 2022-03-11:
How do you create your environment?
Try this first to see if there is any problem.https://mne.tools/stable/install/index.html
```
conda create --name=mne --channel=conda-forge mne
```

RuxueG, 2022-03-11:
This is exactly what I did in the beginning. At the first time, I got information as below:

Platform:       Windows-10-10.0.22000-SP0
Python:         3.9.10 | packaged by conda-forge | (main, Feb  1 2022, 21:22:07) [MSC v.1929 64 bit (AMD64)]
Executable:     F:\anaconda\anaconda\envs\mne\python.exe
CPU:            Intel64 Family 6 Model 142 Stepping 10, GenuineIntel: 8 cores
Memory:         7.9 GB

mne:            0.24.1
numpy:          1.22.3 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:          1.8.0
matplotlib:     3.5.1 {backend=module://matplotlib_inline.backend_inline}

sklearn:        1.0.2
numba:          0.53.1
nibabel:        3.2.2
nilearn:        0.9.0
dipy:           1.4.1
cupy:           Not found
pandas:         1.4.1
mayavi:         4.7.2
pyvista:        0.33.3 {OpenGL 4.5.0 - Build 25.20.100.6576 via Intel(R) UHD Graphics 620}
pyvistaqt:      0.7.0
ipyvtklink:     Not found
vtk:            9.0.3
PyQt5:          5.12.3
ipympl:         Not found
mne_qt_browser: 0.2.5
pooch:          v1.6.0

Then I checked every package and tried to update them to the latest version as shown in my comment #2. But none of the two settings solve the problem in the 'brain.add_volume_labels()'. 

Before the error prompt up, there's a warning:
F:\anaconda\anaconda\envs\mne\lib\site-packages\mne\viz\backends\_pyvista.py:192: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(pyvista.__version__) >= '0.27.0':

I was wondering if this warning matters. And how should I deal with it? 

Thank you for your help!

larsoner, 2022-03-11:
In this case I think it's because MNE is out of date. Can you try:
```
conda remove -f mne
pip install https://github.com/mne-tools/mne-python/zipball/main
```
And see if it helps?

larsoner, 2022-03-11:
(... out of date in the sense that it's installing the stable version, and to deal with this particular problem you need the dev version. FYI, the next MNE version should come out next week!)

RuxueG, 2022-03-11:
Thank you! It works now!

Looking forward to the new MNE version!

closed 2022-08-31 by larsoner