mne-tools/mne-python #10428: support latin1 encoding for channels in read_raw_nihon

MatthiasEb, 2022-03-10:
When opening Nihon-Kohden files with `read_raw_nihon` I get the error

```
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc4 in position 1070: invalid continuation byte
```
raised in [this line](https://github.com/mne-tools/mne-python/blob/1b14d458cbcd391a6907032b809ec37410b60592/mne/io/nihon/nihon.py#L111).

I guess it is similar to [this issue](https://github.com/mne-tools/mne-python/issues/9267) which was already solved [here](https://github.com/mne-tools/mne-python/pull/9384), as replacing the above mentioned line with 
```
with open(e_fname, 'r', encoding='latin1') as fid:
```

fixed the issue.

Maybe we could add a similar [loop over different encodings](https://github.com/mne-tools/mne-python/pull/9384/files#diff-800fa074fc54d9bab7e81ae7d9f3239321fdf53d127bf6bbea3935a3095586e5R252) in [_read_21e_file](https://github.com/mne-tools/mne-python/blob/1b14d458cbcd391a6907032b809ec37410b60592/mne/io/nihon/nihon.py#L102)?

cbrnr, 2022-03-10:
No objection, can you make a PR?

MatthiasEb, 2022-03-10:
I'll have to see how this works, but I'll try

PR #10429 (2022-03-10): FIX: support latin1 in channels in nihon files
Fixes #10428 similar to [this PR](https://github.com/mne-tools/mne-python/pull/9384)
closed 2022-03-10 by agramfort