mne-tools/mne-python #10435: Time of acquisition from info['meas_date'] is not always accurate

jhouck, 2022-03-14:
#### Describe the bug
For some fif files, depending on the settings of the DACQ computer, it is possible for the acquisition time listed in info['meas_date'] to be inaccurate. In our case in data over several years, even though the times reported in both FIFF.FIFF_FILE_ID and FIFF.FIFF_MEAS_DATE are labeled as MDT/MST, FIFF.FIFF_MEAS_DATE is consistently 6-7 hours earlier. We have some MEG calendar data for this time and the time from FIFF.FIFF_FILE_ID is accurate. It seems like FIFF.FIFF_MEAS_DATE was in UTC despite reporting as MDT/MST.

#### Steps to reproduce
# Download file https://www.dropbox.com/s/jlxkxs95qqlrktv/noise_run_raw.fif?dl=0
# and place it in a convenient directory, like /tmp
import mne
import pytz
fname_raw = '/tmp/noise_run_raw.fif'
info = mne.io.read_info(fname_raw)
# convert time stamp to datetime using hidden function
new_meas_date = mne.utils._stamp_to_dt((info['file_id']['secs'], info['file_id']['usecs']))
time_diff = new_meas_date - info['meas_date']
print(time_diff)
print(info['meas_date'].astimezone(pytz.timezone('US/Mountain')).strftime('%X %x'))
print(new_meas_date.astimezone(pytz.timezone('US/Mountain')).strftime('%X %x'))

#### Expected results
Times should be very close together

#### Actual results
Times are about 6.5 hours apart

#### Additional information
It might be good to return a warning if the timedelta between the times in FIFF.FIFF_MEAS_DATE & FIFF.FIFF_FILE_ID exceeds some minimum threshold (1 hour?). It is not clear how common this is or how often accurate scan time information is needed. I think FIFF.FIFF_MEAS_DATE is the time that either the 'GO!' button is pressed or 'Record raw' is checked, so when possible this would probably be the preferred time.

Platform:       Linux-5.4.0-81-generic-x86_64-with-glibc2.27
Python:         3.9.2 | packaged by conda-forge | (default, Feb 21 2021, 05:02:46)  [GCC 9.3.0]
Executable:     /export/research/analysis/human/jhouck/shared/tools/anaconda/envs/mne24dev2/bin/python
CPU:            x86_64: 128 cores
Memory:         1007.8 GB

mne:            1.0.dev0
numpy:          1.20.2 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:          1.6.3
matplotlib:     3.4.2 {backend=Qt5Agg}

sklearn:        0.24.2
numba:          0.53.1
nibabel:        3.2.1
nilearn:        0.7.1
dipy:           1.4.1
cupy:           Not found
pandas:         1.2.4
pyvista:        0.32.1 {OpenGL 3.3 (Core Profile) Mesa 20.0.8 via llvmpipe (LLVM 10.0.0, 128 bits)}
pyvistaqt:      0.6.0
ipyvtklink:     0.2.1
vtk:            9.0.1
PyQt5:          5.12.3
ipympl:         Not found
pooch:          v1.5.2

mne_bids:       Not found
mne_nirs:       Not found
mne_features:   Not found
mne_qt_browser: Not found
mne_connectivity:Not found

larsoner, 2022-03-15:
Can you share a file (preferably unaltered / straight from the acq machine) plus the date/time you know the acquisition started?

I have a file I know (based on my calendar) I recorded on 2014/03/12. Looking at that file:
```
>>> info['meas_date']
datetime.datetime(2014, 3, 12, 18, 47, 48, 314988, tzinfo=datetime.timezone.utc)
>>> info['utc_offset']
>>> info['file_id']['secs']
1394650069
>>> mne.utils._stamp_to_dt((info['file_id']['secs'], info['file_id']['usecs']))
datetime.datetime(2014, 3, 12, 18, 47, 49, 892115, tzinfo=datetime.timezone.utc)
```
So these two are consistent to the `sec` level. I'll see if I can find a newer file that hopefully has the `['utc_offset']` set properly...

> I think FIFF.FIFF_MEAS_DATE is the time that either the 'GO!' button is pressed or 'Record raw' is checked, so when possible this would probably be the preferred time.

We would have to ask the MEGIN folks since I'm not sure. IIRC it's earlier than "record raw", though, because the file has already started at that point, and when you hit "Record raw" the `first_samp` gets set.

Locally to diagnose this, if you have access to the scanner, try doing a 10 sec recording where you do all steps quickly, and see if they match. Then try separating some of the steps. (For example, checking the "Record raw" shouldn't affect the file_id <-> meas_date relationship, it should just make first_samp larger.)

Also, are you certain that the file(s) you are looking at are directly from the DAQ, and not, say, opened in MNE-Python and then saved out again? IIRC this will change the `file_id` to something else that reflects when the new file was written, while `meas_date` should stay the same as it was in the original acquisition file.

jhouck, 2022-03-15:
> are you sure the time-zone information is present in the fif file? do you know which fif tag is used? Message ID: ***@***.***>
> […](#)

I think it's in UTC and then show_fiff/mne_show_fiff report it using the offset from the local machine.  For me, the sample data sample_audvis_raw time is reported in MST (i.e, 3 Dec 2002 @ 12:01:10 pm MST).

jhouck, 2022-03-15:
> Can you share a file (preferably unaltered / straight from the acq machine) plus the date/time you know the acquisition started?
>
> I have a file I know (based on my calendar) I recorded on [...]

Unfortunately I don't know the acquisition time of the empty room file that is linked above.  If you don't mind a larger file I can upload something with a known scan time.  Maxfilter adds a new `meas_time` (tag 204) but the original time is also retained. The info will always report a single `meas_time` -- I believe the first one encountered in the fif header -- but the others are also present.

larsoner, 2022-03-16:
Okay I found a file from our newer acq:
```
>>> info['meas_date']
datetime.datetime(2022, 3, 8, 22, 23, 51, 556973, tzinfo=datetime.timezone.utc)
>>> info['utc_offset']
'-08:00'
>>> mne.utils._stamp_to_dt((info['file_id']['secs'], info['file_id']['usecs']))
datetime.datetime(2022, 3, 8, 22, 23, 50, 406074, tzinfo=datetime.timezone.utc)
```
This all seems reasonable to me.

I'm guessing that there is something about how your site did that acquisition that gives it the behavior you see. Perhaps they started the initial parts of the acq process really (hours) early, and this is what determines the `meas_date`. Then they later hit GO, which starts the actual file writing and the `file_id` gets set. (Then they click "record raw", which sets the `first_samp`.)

I would experiment a bit with some acquisition processes to see if you can confirm this. If you do everything as quickly as possible, I bet the dates match better than if you wait some minutes between steps...

larsoner, 2022-03-16:
Indeed this sounds like a mis-configuration between kaptah/sinuhe -- might be worth contacting MEGIN to see if they can fix it.

In the meantime, you should be able to use this to fix your problematic files on read:

https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.set_meas_date

larsoner, 2022-03-16:
... closing since I think there is nothing to change at the MNE for this, but we can reopen if set_meas_date is not enough!

closed 2022-03-16 by larsoner