mne-tools/mne-python #10437: Channels marked as "bad" have further specification

sportnoah14, 2022-03-16:
Hi, 

I presented this idea at a previous office hours. I work in a lab that records ieeg data from epilepsy patients. Due to the nature of the data, apart from some channels having a poor signal quality there are some of channels that show interictal activity (sharp spikes that confound just about any analysis), are seizure onset zones, are outside of the brain or surrounded by damaged tissue such as a lesion. My group likes to apply one or more of these tags to channels when they apply as they assist in analyzing and interpreting this data. My initial thought, and that of someone else, was to place this information in an `Annotation` object like so:

```
sozChns = ['RDa1','RDa2']
spikeyChns = ['RDa1','RDa2','LDh1']
outChns = ['RDh16','LIp16']
chan_annots = Annotations(
    onset = [0,0,0,0], # Onset and duration indicate that these annotations apply to the entire block and not a section
    duration = [0,0,0,0],
    description = ['soz','interictal','out'],
    ch_names = [ sozChns,spikeyChns,outChns ]
)
```

We can then aggregate all these channels into the `info['bads']` list but it's useful, especially when looking at something like seizure networks, to be able to label channels using these other tags. I've also considered adding this information to `_raw_extras` dictionary in the `Raw` object but it wouldn't survive an io round trip. It can be saved in a separative text file (maybe that's how it's done using BIDS) but it's more convenient if it can be saved within the object. 

To summarize this long post, it would be nice to be able to apply tags of some kind to channels.

larsoner, 2022-03-16:
We have considered adding generic channel tags in the past, but settled on only needing "spatial tags" (e.g., MEG sensor triplets that all share the same x/y/z pos but different orientations and/or coil types) that are usually related just to the hardware acquisition setup. So adding this stuff to `info` perhaps doesn't make sense.

Adding it to Annotations, as you suggest, could work. One way would be to allow `onset` and `duration` to be `np.nan` and/or `np.inf`. This would be one easy way (code-wise) to say "always true for this channel". We'd have to update all our plotting code to deal with these somehow, but it would be doable.

In the meantime, setting `onset` to zero and `duration` to `raw.times[-1]` or something might get you something that at least shows up in the plots and survives I/O round trip.

agramfort, 2022-03-16:
+1 to use np.info or raw.times[-1] for duration

it seems easy enough

> Message ID: ***@***.***>
>