mne-tools/mne-python #10445: MAINT: Replace our internal STFT with scipy.signal.stft

larsoner, 2022-03-18:
Once https://github.com/scipy/scipy/pull/15704 is merged we should ideally be able to directly call STFT and preserve power. However, that would require waiting ~2 years. Instead I think we should add a version to `fixes.py` where the appropriate scale factors from https://github.com/scipy/scipy/pull/15704 are used to preserve power.

hoechenberger, 2022-03-19:
> Once [scipy/scipy#15704](https://github.com/scipy/scipy/pull/15704) is merged

Seems it was merged shortly after you opened this issue!

larsoner, 2022-03-19:
I took a quick look yesterday and this is still a pain, at the very least because we pad at the beginning and end while scipy just pads at the end. But looking again at [scipy.signal.stft](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.stft.html) if we keep our internal padding and pass `boundary=None, padded=False` it might work. I'll look again soon...

larsoner, 2023-10-01:
We should try establishing equivalence with (or just living with if it's better) this: https://scipy.github.io/devdocs/reference/generated/scipy.signal.ShortTimeFFT.html#scipy.signal.ShortTimeFFT . @agramfort maybe a good one for you during the sprint :)