mne-tools/mne-python #10447: add skip_by_annotation to notch_filter and resample

jasmainak, 2022-03-20:
We have TMS-EEG data at MGH for which we're trying to figure what is the best processing pipeline. The TMS pulses are recorded as triggers, so we can create annotations of 10 ms around those triggers and use that to exclude from operations such as ICA and [raw.filter](https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.filter). This seems to work well but we noticed that `skip_by_annotation` is not supported in `notch_filter` and `resample`. Would it make sense to add these functionality to these functions as well? It would also make the API more consistent.

larsoner, 2022-03-21:
Agreed this would be useful!

jasmainak, 2022-12-19:
Thinking about this more, if `resample` function was to have `skip_by_annotation`, how is the "skipped" segment to be handled? Would there still be decimation but no filter since MNE-Python only supports a single sampling frequency ...

larsoner, 2022-12-20:
> how is the "skipped" segment to be handled? Would there still be decimation but no filter since MNE-Python only supports a single sampling frequency ...

Yes there would be no need, it would just get subsampled somehow. An [`upfirdn` (with no filter / a delta)](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.upfirdn.html) could do it as you suggest