mne-tools/mne-python #10451: Make it such that our installer download always point to the correct version and don't need to be updated with each release

hoechenberger, 2022-03-21:
How I imagine this to work:

1. we follow a strict naming / tagging scheme on `mne-tools/mne-installers`
1. we add a tiny snippet of JavaScript to the HTML file of our docs that renders the download buttons; the script extracts the version number of the currently-viewed documentation and adjusts the link targets of the download buttons accordingly
1. profit!

drammock, 2022-06-15:
what about https://github.com/mne-tools/mne-installers/releases/download/**latest**/MNE-Python-**latest**-Linux.sh that is an alias for the most recent version? and you update that link each time you make a release?

hoechenberger, 2022-06-15:
I think that's what we're already doing, unless I misunderstood your proposal … 
And it might just be good enough. I mean, we don't release **that** often, so…

drammock, 2022-06-15:
no, I mean literally create a path https://github.com/mne-tools/mne-installers/releases/download/latest/MNE-Python-latest-Linux.sh that doesn't result in a 404

drammock, 2022-06-15:
https://docs.github.com/en/rest/releases/releases#get-the-latest-release

drammock, 2022-06-15:
https://api.github.com/repos/mne-tools/mne-installers/releases/latest gets you the JSON

larsoner, 2023-10-01:
If we do this with JavaScript it means that for example we do it for 1.5, and when we release 1.6 the 1.5 doc installers will point to 1.6 (because they'll always point to the latest release). I actually think this is okay because the HTML page will display which installer is being talked about (1.6) and it's the only one we'll support at the release of 1.6 anyway. But it's something to be aware of.

The alternative would be to do some version parsing to figure out which version of the docs you're in, then go through the releases using GitHub APIs to figure out which release to use. It's doable but onerous for little gain. Perhaps what we *could* do is detect if the current version of the displayed docs (e.g., 1.5 *or* dev) does not match that of the installers being advertised (1.6) and show a warning. This has the added benefit that people looking at the dev page won't errantly expect to download the installer from there and immediately get a dev version.

drammock, 2023-10-03:
> If we do this with JavaScript it means that for example we do it for 1.5, and when we release 1.6 the 1.5 doc installers will point to 1.6 (because they'll always point to the latest release). I actually think this is okay

agree that this is the best approach. Cases where users will want something other than the latest installer are rare (so far to my knowledge it's only ever happened when one of the deps we bundled was buggy).