mne-tools/mne-python #10453: ContainsMixin doesn't work properly

mscheltienne, 2022-03-22:
I noticed this on the latest version (made a fresh env/pip install since mine was a bit old).

```
In [1]: from mne import create_info
In [2]: info = create_info(ch_names=['Fp1', 'Fp2', 'Fpz'], sfreq=1, ch_types='eeg')
In [3]: 'eeg' in info
Out[3]: False
```

Probably because `__contains__` gets resolved with the inheritance from `dict` before the inheritance from `ContainsMixin`.

Also from the same Mixin, `compensation_grade()` raises this error when there are no MEG channels. I guess it should raise, but something more explicit than this obscure error.

```
In [4]: info.compensation_grade()
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [4], in <cell line: 1>()
----> 1 info.compensation_grade()

TypeError: 'NoneType' object is not callable
```

larsoner, 2022-03-22:
The `__contains__` *should* resolve to `dict` for backward compat. I have code (and I imagine others do, too) that uses things like `if 'subject_info' in raw.info:` etc. and we don't want to break it.

> Also from the same Mixin, compensation_grade() raises this error when there are no MEG channels. I guess it should raise, but something more explicit than this obscure error.

Is `info.compensation_grade` a property? If so, it should return None or an int, and by doing `info.compensation_grade()` (note the extra parens) you are going to be calling `None.__call__` or `int.__call__`, neither of which exists so the error seems okay

mscheltienne, 2022-03-22:
This is why I posted this as an issue and not a PR. I wasn't sure if there was something to fix. So for `Info`, you consider that `__contains__` from `ContainsMixin` should be ignored in favor of the one from `dict`.

For the second, that's a function: https://github.com/mne-tools/mne-python/blob/main/mne/io/meas_info.py#L262-L265
Which points to https://github.com/mne-tools/mne-python/blob/cd5a5608d2183efdab2625627584913cdeddbba6/mne/io/compensator.py#L7-L19
Which returns None if there are no MEG channels. I think a simple update of the error raised with a better message would be enough here.

larsoner, 2022-03-22:
> For the second, that's a function

Looks like you missed the critical `@property` one line above

https://github.com/mne-tools/mne-python/blob/cd5a5608d2183efdab2625627584913cdeddbba6/mne/io/meas_info.py#L261-L265

larsoner, 2022-03-22:
... being a property it means you should use `info.compensation_grade` to get the result, not `info.compensation_grade()`, unless you want to `__call__` the resulting None or int that `info.compensation_grade` (note: no parens!) already returns

mscheltienne, 2022-03-22:
... time for me to shut down the computer and go to bed 🙈 🙈 thanks.

closed 2022-03-22 by mscheltienne

larsoner, 2022-03-22:
Hopefully `@property` will now show up in dreams not nightmares :)