mne-tools/mne-python #10454: annotation window does not close properly

jasmainak, 2022-03-23:
#### Describe the bug
when doing `raw.plot()`, the annotation window does not close even if the main browser is closed. Attempting to close that window throws a spurious error:

```py
In [8]: 

Traceback (most recent call last):
  File "/Users/mainak/anaconda3/envs/mne/lib/python3.9/site-packages/matplotlib/cbook/__init__.py", line 270, in process
    func(*args, **kwargs)
  File "/Users/mainak/Documents/github_repos/mne-python/mne/viz/_mpl_figure.py", line 180, in _close
    super()._close(event)
  File "/Users/mainak/Documents/github_repos/mne-python/mne/viz/_mpl_figure.py", line 103, in _close
    self.mne.parent_fig.mne.child_figs.remove(self)
ValueError: list.remove(x): x not in list
```

#### Steps to reproduce
```py
import os
import numpy as np
import mne

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_filt-0-40_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)
raw.plot()
```

Now press 'a' to open the annotation window, then draw an annotation, then attempt to close the main window. It does not close the annotation window. Next, try to close the annotation window to get the error.

#### Expected results
The annotation window should close when the main window closes and there should not be any error.

#### Additional information
I am on latest master of MNE-Python

agramfort, 2022-03-23:
with the matplotlib or the qt browser?

> Message ID: ***@***.***>
>

jasmainak, 2022-03-23:
I was using `ipython --matplotlib` but I just checked and the error can be reproduced also on `ipython --matplotlib=qt`

hoechenberger, 2022-03-25:
I think @agramfort wanted to know whether you are using the good old Matplotlib-based GUI or the fancy new one, which is our new default in MNE 1.0

agramfort, 2022-03-25:
I can replicate the pb. I tried to investigate. For me these lines

https://github.com/mne-tools/mne-python/blob/main/mne/viz/_figure.py#L423

should do the job.

@marsipu any idea?

marsipu, 2022-03-25:
Child figures are handled by each browser-backend (matplotlib, qt) individually:

#### matplotlib
https://github.com/mne-tools/mne-python/blob/8fde21e3da598dd53c01cd23e1b5bb24d6ecf6df/mne/viz/_mpl_figure.py#L782
https://github.com/mne-tools/mne-python/blob/8fde21e3da598dd53c01cd23e1b5bb24d6ecf6df/mne/viz/_mpl_figure.py#L103
#### qt
https://github.com/mne-tools/mne-qt-browser/blob/c0a7d32259c13097dbd0f2553725ba7049c83b3e/mne_qt_browser/_pg_figure.py#L1492
https://github.com/mne-tools/mne-qt-browser/blob/c0a7d32259c13097dbd0f2553725ba7049c83b3e/mne_qt_browser/_pg_figure.py#L1529

But since a separate window for annotations by pressing `a` only occurs for the `matplotlib`-backend I think this bug is specific for `matplotlib`.

jasmainak, 2022-03-26:
Oh, I didn't realize the Qt-based browser was also available now. Just gave it a shot and it's super nice!

But I do get a similar error:

```py
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
TypeError: closeEvent() takes 1 positional argument but 2 were given
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
~/anaconda3/envs/mne/lib/python3.9/site-packages/mne_qt_browser/_pg_figure.py in closeEvent(self, event)
   4493             if hasattr(self.mne, 'child_figs'):
   4494                 for fig in self.mne.child_figs:
-> 4495                     fig.close()
   4496                 self.mne.child_figs.clear()
   4497             for attr in ('traces', 'event_lines', 'regions'):

RuntimeError: wrapped C/C++ object of type SettingsDialog has been deleted
Out[1]: <mne_qt_browser._pg_figure.PyQtGraphBrowser at 0x12ef2f550>
```

not quite sure what sequence of operations led to it though. 

Also it doesn't allow me to see the time instant by clicking as in the matplotlib browser

marsipu, 2022-03-26:
@jasmainak Ok, this looks like qt-browser seems to have a bug too related to closing the Settings-Dialog. Maybe you could open an issue about that at [mne-qt-browser](https://github.com/mne-tools/mne-qt-browser/issues)?

jasmainak, 2022-03-29:
humm ... acually that gives a *different* error than the one I got. I posted that on the mne-qt-browser github along with another issue I had. I haven't been able to reproduce the [error above](https://github.com/mne-tools/mne-python/issues/10454#issuecomment-1079533695) again. I'll post back if I manage to do that

hoechenberger, 2022-05-05:
@jasmainak has this meanwhile been resolved?

commit 37a9e558 (2022-06-23): Fix orphaned annot window (#10776)

* DRY _close_event

* DRY nested imports

* STY: use relative imports

* STY: isort imports in _mpl_figure.py

* FIX: workaround for matplotlib #23298

* FIX: closes #10454

* DOC: add docstring

* FIX: compat with older matplotlib versions

* isort imports in mne/viz/tests/test_raw.py

* refactor some test helpers, add test that fails on main

* remove debug cruft
closed 2022-06-23 by drammock