mne-tools/mne-python #10469: Use a `float` default value for `fmin`-param in `mne.time_frequency.psd_array_multitaper`

sondrfos, 2022-03-30:
Hello,
 
Static type checkers infers the expected type of the `fmin`-param for the `mne.time_frequency.psd_array_multitaper`-function to be of type `int` since the default value used is an `int`.

The docstring clearly states that `psd_array_multitaper` supports `fmin`-values of type `float`.

Can we change the default-value of `fmin` to `0.0` to avoid errors in static type checkers?

[image]

https://github.com/mne-tools/mne-python/blob/091da8f01aeeecd7d583ba596cf5a85cd649f192/mne/time_frequency/multitaper.py#L365

cbrnr, 2022-03-30:
I'm not a big fan of complicating code just because static type checkers are not happy. But granted, this is a minimal change.

hoechenberger, 2022-03-30:
We don't supply type annotations or type stubs for MNE-Python yet. Static type checkers therefore make **a lot** of mistakes with this package. I'm against "fixing" things in just a single place. I'd like to start a discussion on how and when we add type annotations to MNE's code. It's IMHO overdue.

cbrnr, 2022-03-30:
Actually, we're already providing type annotations in our docstrings.

hoechenberger, 2022-03-30:
So a transition to proper type annotations should not bother you, then.

cbrnr, 2022-03-30:
> I'm not sure I agree that changing from 0 -> 0.0 is complicating the code.

I would - you need 3 characters instead of 1. I did say that this is a minimal change of course, but you can't argue that it doesn't get more complicated/complex/verbose. However, that's not the point anyway, I think we agree that discussing this on a larger scale makes sense.

sondrfos, 2022-03-30:
> I would - you need 3 characters instead of 1. I did say that this is a minimal change of course, but you can't argue that it doesn't get more complicated/complex/verbose. However, that's not the [...]

Yes, we agree on your final point! Would love a larger scale discussion :)

agramfort, 2022-03-31:
I also tend to write 0. or 0.0 to make my code explicit that float is expected.

+0.5 on fixing this one with a small PR.

@sondrfos would you have time to do this,

regarding type annotations everywhere in MNE let's discuss this over a drink to cool down the conversation ;)

cbrnr, 2022-03-31:
And yes, let's fix this tiny number here.

hoechenberger, 2022-03-31:
@cbrnr 
Your examples are not well chosen. Well, they're chosen to support your point, but they're not good examples per se. 

When using type annotations, one would obviously format the function signatures differently. Typically, one parameter per line. 

Your example would be like tryingtoprovethatleavingoutspacesinwrittentextisabadidea. We all already know that it is.

PR #10479 (2022-03-31): Use a `float` default value for `fmin`-param in `mne.time_frequency.psd_array_multitaper`
Fixes #10469.

This is a really tiny, nitpicky fix. 
If anyone wants more more text here please comment and I will expand the description.
closed 2022-03-31 by agramfort