mne-tools/mne-python #10470: pick_types(csd=True) does not return csd type channels but fails with a ValueError

matthiasdold, 2022-03-30:
Hi all,
I am pretty sure, this is not as it is supposed to be, but would be fixed quite easily.

#### Describe the bug
When working in an instance with [CSD transformed](https://mne.tools/stable/auto_examples/preprocessing/eeg_csd.html?highlight=csd) data, using `.pick_types(csd=True)` returns an error complaining about "no channels match the selection".

This is due to the `csd=True` kwarg not being passed to `pick_types` in according method of `mne.channels.channels.UpdateChannelsMixin` - [see here](https://github.com/mne-tools/mne-python/blob/maint/1.0/mne/channels/channels.py#L671-L676)

#### Steps to reproduce
```python
import mne
from pathlib import Path
from mne.datasets import sample

data_path = sample.data_path()
meg_path = Path(data_path) / 'MEG' / 'sample'
raw = mne.io.read_raw_fif(meg_path / 'sample_audvis_raw.fif')
raw = raw.pick_types(meg=False, eeg=True, eog=True, ecg=True, stim=True,
                     exclude=raw.info['bads']).load_data()
events = mne.find_events(raw)
raw.set_eeg_reference(projection=True).apply_proj()
raw_csd = mne.preprocessing.compute_current_source_density(raw)

raw_csd.pick_types(csd=True)
```

#### Expected results
The instance object filtered for the 59 CSD transformed channels. See info:
```python
In [22]: raw_csd.info
Out[22]:
<Info | 21 non-empty values
 acq_pars: ACQch001 110113 ACQch002 110112 ACQch003 110111 ACQch004 110122 ...
 bads: []
 ch_names: STI 001, STI 002, STI 003, STI 004, STI 005, STI 006, STI 014, ...
 chs: 9 Stimulus, 59 Current source density, 1 EOG
```
#### Actual results
The error:
```python
ValueError: No channels match the selection.
```

agramfort, 2022-03-31:
thanks for the detailed bug report. WOuld you have time to submit a PR to
fix this?

thanks

> Message ID: ***@***.***>
>

matthiasdold, 2022-04-01:
PR is up. I hope, I did everything correctly. But should really minor anyways.

PR #10484 (2022-04-03): MRG FIX: csd=True kwarg to be passed to pick_types

#### Reference issue
Fixes #10470.

#### What does this implement/fix?
Added `csd=csd` to `mne.channel.channels.UpdateChannelsMixing.pick_types()` method. The kwarg was not passed to the call to `pick_types`.

Now using `.pick_types(csd=True)` on an instance object like `raw.io.BaseRaw`properly returns the csd type channels.

With this PR we get:
```python
import mne
from pathlib import Path
from mne.datasets import sample

data_path = sample.data_path()
meg_path = Path(data_path) / 'MEG' / 'sample'
raw = mne.io.read_raw_fif(meg_path / 'sampl [... cut]
closed 2022-04-03 by larsoner