mne-tools/mne-python #10481: Use matplotlib-based raw browser in docs

cbrnr, 2022-04-01:
I think that showing raw plots in our docs should not include toolbar icons, scrollbars, and other interactive (irrelevant) items, as is the case since 1.0 (e.g. the screenshot in https://mne.tools/dev/auto_examples/io/read_xdf.html).

I would switch to the matplotlib backend in our docs to get rid of UI elements, unless it is also possible to do this with the qt backend (zen mode maybe?).

larsoner, 2022-04-01:
> I would switch to the matplotlib backend in our docs

Looks like you missed https://github.com/mne-tools/mne-python/pull/9960 and https://github.com/mne-tools/mne-python/pull/10359 where we worked out using the Qt backend in the doc build, since this sounds like an argument to undo most of that. Given that people at the time at least were more or less "pro" this change IIRC, I'd assume they're -1 on reverting it now.

> unless it is also possible to do this with the qt backend (zen mode maybe?).

In principle we could add zen-mode-style capture. However, even back when we used matplotlib we included the UI elements like scrollbars, e.g. [here](https://mne.tools/0.24/auto_tutorials/intro/15_inplace.html#sphx-glr-auto-tutorials-intro-15-inplace-py). Personally I don't mind having them in there because it's a bit closer to the user experience and doesn't take up too much space (unlike in `Brain` where the UI elements arguably do, and we do not capture them). Also, the scrollbars do convey implicit information (e.g., number of channels). So I'm -1 on a full "zen mode" where we remove all interactive elements including scrollbars and overview mode (not sure if that was included in your concept or not).

We could consider individual elements:

1. The toolbar doesn't add information and might be easy to hide, so we could do that.
2. Scrollbars are small, so removing them isn't a huge gain. We could probably `setVisible(False)` them but I don't think it will change much (and again, loses a bit of information).
3. The overview bar is at least sometimes helpful/informative, so it's nice to have it in there. In individual examples/tutorials we could also switch it to z-score mode if it would help. So I'd rather not remove that.
4. "Overview bar mode" selector -- actually I wonder if we should move this up to the toolbar (@marsipu ?). It's a better use of vertical space, even though it moves it away from the overview. Given how rarely it's changed, it seems worth it. I'll open a mne-qt-browser issue to discuss separately...

I guess we could hide the toolbar and overview-mode-selector. The downside is that it creates a mismatch with what the user will see, but I guess it's an understandable one. I'm +/-0 on this.

cbrnr, 2022-04-01:
> Looks like you missed https://github.com/mne-tools/mne-python/pull/9960 and https://github.com/mne-tools/mne-python/pull/10359 where we worked out using the Qt backend in the doc build, since this sounds like an argument to undo most of that.

Seems like I did!

I'm OK with not hiding everything, but at least the toolbar and the overview bar do not add any important information. If they do in a particular example, we could explicitly enable them, but by default I'd hide them.

Usually, we want to show the continuous signals in the tutorials (even in the example you mentioned), and we don't need the scrollbars or anything else.

So if the backend cannot be reverted, I'd vote for a zen-mode-like display (no overview bar, no toolbar, and even no scrollbars, although I'd be OK with leaving those in).

cbrnr, 2022-04-04:
I'd love an `overview_mode=...` argument. I think we should expose all browser settings as arguments as a general rule.

I would leave the overview bar on for users, because it adds helpful information. However, I think the overview bar should be opt-in in the docs. Usually, we want to show the raw signals and not so much the exact interactive interface. If this is the goal, we can enable the bar explicitly.

I'd actually prefer the argument over the scraper, because I think it might be useful for people to open the raw browser with their preferred overview bar mode. This could also be a setting we could store in the config.

larsoner, 2022-04-04:
That would end up being pretty easy implementation-wise, too, because we can just `export MNE_BROWSER_OVERVIEW_MODE=off` or false or whatever in CircleCI. I can start on an implementation for this in the coming days

PR #10501 (2022-04-11): ENH: Add overview_mode support
1. Add `overview_mode` / `MNE_BROWSER_OVERVIEW_MODE` config var to control which overview bar is shown
2. Removes the overview bar when generating docs. We should find examples where we want to keep it (probably some with annot/events), but for now let's just disable it.

Closes #10481
closed 2022-04-11 by agramfort