mne-tools/mne-python #10482: Missing event information when reading EEGLAB file and exporting raw to .set files

cmadjar, 2022-04-01:
#### Describe the bug

When reading an EEGLAB file with `mne.io.read_raw_eeglab()`, the events information seems oversimplified.

In EEGLAB,  the set file event structure contains the following:
```
>> EEG_set.event

ans = 

  1×2670 struct array with fields:

    description
    begintime
    classid
    code
    duration
    label
    relativebegintime
    sourcedevice
    name
    tracktype
    latency
    type
    mffkeys
    mffkey_gidx
    mffkeysbackup
    mffkey_age_
    mffkey_exp_
    mffkey_hand
    mffkey_sex_
    mffkey_subj
    mffkey_cel
    mffkey_obs
    mffkey_spc
    mffkey_pos
    mffkey_argu
    mffkey_rsp
    mffkey_eval
    mffkey_rtim
    mffkey_trl
    mffkey_blk1
```

However, after reading the SET file with `mne.io.read_raw_eeglab()`, I can only see annotations with `onset`, `duration`, `description` and `orig_time` information:
```
>>> raw.annotations
<Annotations | 2670 segments: CELL (4), DIN2 (666), SESS (1), TRSP (666), ...>
>>> raw.annotations[0]
OrderedDict([('onset', 0.255321), ('duration', 2.0), ('description', 'SESS'), ('orig_time', None)])
>>> 
>>> raw.annotations[1]
OrderedDict([('onset', 0.46732), ('duration', 2.0), ('description', 'CELL'), ('orig_time', None)])
```

And when I export the raw structure into a new SET file, the event structure of the file is as follows:
```
>> EEG_new_set.event

ans = 

  1×2670 struct array with fields:

    type
    latency
    duration
```

Is there a way to keep the same events structure as the original SET file?

#### Steps to reproduce

```
raw = mne.io.read_raw_eeglab(input_fname=<FILE PATH>, preload=False, verbose=True)
mne.export.export_raw(
    fname=<NEW FILE PATH>,
    raw=raw,
    fmt='eeglab',
    overwrite=True
)
```

#### Additional information

```
>>> mne.sys_info()
Platform:         macOS-11.3-x86_64-i386-64bit
Python:           3.9.12 (main, Mar 26 2022, 15:52:10)  [Clang 13.0.0 (clang-1300.0.29.30)]
Executable:       /Users/cmadjar/Development/python_venv/python_intel_3.9.4/eeg2bids_py3.9.12/bin/python
CPU:              i386: 8 cores
Memory:           Unavailable (requires "psutil" package)
mne:              1.0.0
numpy:            1.21.5 {blas=openblas, lapack=openblas}
scipy:            1.8.0
matplotlib:       3.5.1 {backend=MacOSX}

sklearn:          1.0.2
numba:            0.55.1
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
cupy:             Not found
pandas:           1.4.1
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
PyQt5:            Not found
ipympl:           Not found
pooch:            v1.6.0

mne_bids:         0.10
mne_nirs:         Not found
mne_features:     0.2
mne_qt_browser:   Not found
mne_connectivity: Not found
```

agramfort, 2022-04-01:
MNE cannot read arbitrary fields in the event structure. If this information can be store in an annotations object let's fix it but otherwise we need to see. What do these fields contain?

SMoralesPhD, 2022-04-04:
Those fields contain details about the conditions of interest. For example, for a face processing task, we would like to code the following attributes: upright vs inverted, gaze towards vs away, race of the face, trial number, block number, etc. Is there a way to add that information to the event structure? 
From reading the documentation, it seems like it would be possible: "An arbitrary number of additional columns can be added. Those allow describing other properties of events that could be later referred in modeling and hypothesis extensions of BIDS." 
Thanks!