mne-tools/mne-python #10483: concatenate raws and keep the temporality

etiennedemontalivet, 2022-04-01:
#### Describe the new feature or enhancement
Allow users to keep the temporality when concatenating multiple ``Raw``

#### Describe your proposed implementation
Either by improving the current ``mne.concatenate_raws`` with a flag or by creating a new function (where ?) that would fill the gap with 0s ? and annotate them with either ``GAP`` or ``BAD_ACQ_SKIP`` ?

#### Describe possible alternatives
We could also provide a non-continous ``Raw.times`` ? But I am not sure it won't break anything else

#### Additional comments
I think it would be very interesting because a lot of EEG recording sessions are made of multiple recordings due to electronic issues, software issues,...

I would be glad to integrate it, I would just need some directions. Thanks !

agramfort, 2022-04-01:
are big are the gaps? do you foresee memory issues if we fill with zeros or
NaNs?

> Message ID: ***@***.***>
>

agramfort, 2022-04-03:
one issue I see is that Raw offers convenient methods on top of the
raw._data attribute that is just one numpy array.
Changing this so we now have eg a list of numpy arrays for each segment of
data would have lots of consequences
on a lot of code. So to it's not really a problem of Annotations but more
how data are stored and accessed internally.

Message ID: ***@***.***>
>

etiennedemontalivet, 2022-04-04:
I agree with that. I don't think the ``raw._data`` attribute has to be changed at all. I was suggesting two approaches:

1. simply fill gaps with ``0``. The data structured won't be changed. It could however leads to memory issue, so I suggested to annotate these ``0`` with ``BAD_ACQ_SKIP`` so it not stored in memory. 
2. change the ``raw.times`` so that the time is discontinuous. I am less confortable with this one as I think I don't see all the consequences of such a change, but I just wanted to mention it as it seems doable and you might prefer it for reasons I don't know.

The final idea is to be able to concatenate recordings and to keep the **temporality** so one can easily use external time annotations to annotate the concatenation.

A **real-life example** where this feature would be a plus is in Epilepsy research. The ``chb-mit`` database contains multiple recordings for each subject that are separated with small gaps (~few seconds). See [here](https://physionet.org/content/chbmit/1.0.0/chb01/chb01-summary.txt). One would definitely love having one fif-file for each subject, and then annotate them with the provided ``datetime``.

**Code** for solution 1 would look like:

```python
def concatenate_and_fill_raws(mne_raws: list) -> mne.io.RawArray:
    info = mne_raws[0].info

    # Check that filling gaps is possible
    for iRec, raw in enumerate(mne_raws):
        if raw.info["sfreq"] != info["sfreq"]:
            raise ValueError(
                f"Cannot fill gaps if Raws have different frequencies.(0:{info['sfreq']}" + 
                f" - {iRec}:{raw.info['sfreq']})"
            )
        if raw.ch_names != mne_raws[0].ch_names:
            raise ValueError(
                f"Cannot fill gaps if Raws have different channel names.(0:{mne_raws[0].ch_names}" + 
                f" - {iRec}:{raw.ch_names})"
            )
        if iRec > 0:
            if (
                not mne_raws[iRec - 1].info["meas_date"]
                < mne_raws[iRec].info["meas_date"]
            ):
                raise ValueError(
                    "mne_raws list must be ordered regarding measurement dates."
                )

    # concatenate recordings and fill gaps
    concat_raws = mne_raws[0]
    for iRec, raw in enumerate(mne_raws):
        if iRec > 0:
            # we must fill the gap between the current and the previous
            previous_raw = mne_raws[iRec - 1]
            diff_in_sec = (
                raw.info["meas_date"]
                - (
                    previous_raw.info["meas_date"]
                    + timedelta(0, previous_raw.times[-1])
                )
            ).total_seconds()
            n_samples_to_add = int(diff_in_sec * info["sfreq"])

            if n_samples_to_add < 0:
                raise ValueError(f"Overlapping raws cannot be correctly concatenated")

            if n_samples_to_add > 0:
                gap = mne.io.RawArray(np.zeros((info["nchan"], n_samples_to_add)), info)
                gap.annotations.append(
                    onset=0,
                    duration=n_samples_to_add / info["sfreq"],
                    description="BAD_ACQ_SKIP",
                )
                concat_raws.append(gap)
            concat_raws.append(raw)

    return concat_raws
```
Code has not been properly tested but seems to work

agramfort, 2022-04-07:
I don't see who using a BAD_ACQ_SKIP Annotation would fix the
memory problem. You would still have to store the zeros AFAIK

> Message ID: ***@***.***>
>

etiennedemontalivet, 2022-04-08:
It is a specific behaviour of ``BAD_ACQ_SKIP`` annotation. Larsoner told me that in [this issue](https://github.com/mne-tools/mne-python/issues/9985#issuecomment-964202216). There is now some doc about it [here](https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.save).

agramfort, 2022-04-08:
right but it would keep the zeros in RAM with preload=True

I understand that on disk no

I did not think deeply about this...

> Message ID: ***@***.***>
>

etiennedemontalivet, 2022-04-08:
Yes, you're right. I was definitely thinking about short period gaps cases.

agramfort, 2022-04-08:
+1 to see what we can do for short gaps

> Message ID: ***@***.***>
>