mne-tools/mne-python #10486: Argument `theme` of `raw.plot()` does not work

mscheltienne, 2022-04-02:
As spotted following the discussion here https://github.com/mne-tools/mne-qt-browser/issues/83 the argument `theme` does not change the theme on either my macOS or Windows computer.

Sample code:

```
import mne

folder = mne.datasets.sample.data_path() / 'MEG' / 'sample'
raw = mne.io.read_raw(folder / 'sample_audvis_filt-0-40_raw.fif', preload=True)
raw.plot(theme='dark')
```

mscheltienne, 2022-04-07:
@larsoner can you re-open this one?
I am still getting a dark theme no matter the `theme` argument on my macOS with dark-mode enabled a the system-level.
Setting it in the MNE config with `mne.set_config('MNE_BROWSER_THEME', 'light')` does not fix it either.

mscheltienne, 2022-04-07:
I don't think so. I can give it a shot with the latest version of `mne-qt-browser` later; but in my case, the UI stays in dark mode no matter what. The entire UI.

mscheltienne, 2022-04-07:
Fresh environment with both `mne-python` and `mne-qt-browser` installed from `main`. No luck, there is something wrong here.

```
folder = mne.datasets.sample.data_path() / 'MEG' / 'sample'
raw = mne.io.read_raw(folder / 'sample_audvis_filt-0-40_raw.fif', preload=True)
raw.plot(theme='light')
```

[image]

larsoner, 2022-04-07:
Can you investigate a bit? If not, I can try to look in a bit

mscheltienne, 2022-04-07:
I can look at it tomorrow.

PR #10487 (2022-04-03): MRG: Fix passing of argument theme for raw.plot()
Fix #10486 on my windows computer.
It probably also fixes it on my macOS (can't remember if I did set the config variable or not..), but I can't test until Monday.
PR #10500 (2022-04-10): BUG: Fix bug with themes on macOS
@hoechenberger @mscheltienne . This code:
```
python -c "import mne; raw = mne.io.read_raw_fif(mne.datasets.sample.data_path() / 'MEG' / 'sample' / 'sample_audvis_raw.fif'); raw.plot(theme='auto', block=True)"
```
Gives

| macOS mode | `theme='auto'` | `theme='dark'` | `theme='light'` |
| -- | -- | -- | -- |
| dark | [image] | [image] | [image] | 
| light | [image] | [image] | [image] |

Fixes #10486
Fixes https://github.com/mne-tools/mne-qt-browser/issues/100

The one real downsid
closed 2022-04-10 by agramfort