mne-tools/mne-python #10507: make_inverse_operator() should warn upon mismatch between "info" and empirical "noise_cov" rank

hoechenberger, 2022-04-08:
As per discussion on Discord w/ @agramfort, @larsoner, @britta-wstnr, @alexrockhill

larsoner, 2022-04-08:
We have something that is meant to do this I think here

https://github.com/mne-tools/mne-python/blob/41d38b7a55afeb2a6a840d07a155dae837326ef0/mne/rank.py#L415-L425

From discord it sounds like the bug is really when the rank of the cov is *less* than the rank of the `info['data']`, and this code path isn't even triggered because `rank='info'`.

So maybe what we need is a check in the `rank='info'` path where it *still* computes the rank, and checks to see if the rank from info is probably/likely wrong.

hoechenberger, 2022-04-08:
> and this code path isn't even triggered because `rank='info'`.
> 
> So maybe what we need is a check in the `rank='info'` path where it _still_ computes the rank, and checks to see if the rank from info is probably/likely wrong.

Yes, this sounds reasonable!