mne-tools/mne-python #10524: Micromed TRC file reader

jungju69, 2022-04-13:
Hello,
I have several datasets based on micromed eeg files (scalp EEG, SEEG, ERP).
Do you think it could be possible to develop file readers for that format (TRC files), quite commonly used ? 
thank you again for your nice software !

agramfort, 2022-04-13:
see this old discussion https://github.com/mne-tools/mne-python/issues/1605

@jungju69 would you up for writing such a reader?

jungju69, 2022-04-13:
I would love to … But my python programming skills are too light.
I can convert trc files to edf format and use MNE but I was looking for a more direct approach.
Any other suggestion ?
best
JJ

ATTENTION : Ce message a été envoyé par un expéditeur externe, en dehors des Hospices Civils de Lyon. Ne cliquez pas sur les liens ou n’ouvrez pas les pièces jointes… sauf si vous connaissez l’expéditeur et que vous lui faites confiance !

agramfort, 2022-04-13:
> Any other suggestion ?

see my gist in the  issue I linked to

otherwise find someone to help. Maybe some engineer in your lab?

sappelhoff, 2022-09-01:
or you could find a documentation of micromed's `.trc` file format and post it here, maybe someone in the future will implement this.

mossdet, 2023-10-17:
Hello MNE community,

I also work with Micromed files from time to time. I wrote a reader that uses python3 and numpy, it is based on the fieldtrip Micromed reader. 

I'm not sure how you can integrate this reader into the MNE infrastructure, I could send you the script if this helps.

Done:
- I compared the read data with the MATLAB reader and it's running fine. 
- I can plot the individual channels using matplotlib and this is also running fine. 
- I created an MNE Raw Array and this also works 

ToDo:
- The mne.io.RawArray.plot() is not working for some reason. I don't know if I need to define something else when creating the RawArray from scratch.

larsoner, 2023-10-17:
> I also work with Micromed files from time to time. I wrote a reader that uses python3 and numpy, it is based on the fieldtrip Micromed reader.

Unfortunately as far as I understand it FieldTrip is GPL licensed and you cannot use a GPL implementation (FieldTrip) as a reference when adding an implementation to a BSD package (MNE-Python). You'd have to ask permission from the FieldTrip folks to relicence as BSD.

But if you manage to do that, it does sound like you're close to having a workable solution that we could add!

mossdet, 2023-10-17:
My reader is based on the fieldtrip reader in the sense that I used it to know how the data bytes are organized in the TRC files, I don't know how to get the format structure otherwise (Is Micromed still around as a company?). The reader itself is quite simple and just goes through the file byte by byte sorting the data. I have no experience with Licensing but I would think that it is too simple for there to be any risk of Licensing infringement.

drammock, 2023-10-17:
> I have no experience with Licensing but I would think that it is too simple for there to be any risk of Licensing infringement.

Translations of code from one language to another are specifically mentioned in the GPL. https://www.gnu.org/licenses/gpl-faq.html#TranslateCode

@robertoostenveld one of our users has translated the FieldTrip MicroMed file reader code into Python and is proposing to add it to MNE-Python. Requesting permission to re-license as BSD 3-clause.

larsoner, 2023-10-20:
+1 for using Neo if it has what we want, we are doing something similar for Neuralynx data in #11969 which could be followed as a template

drammock, 2023-10-20:
thanks @robertoostenveld for the info and suggestions.

mossdet, 2023-10-24:
Thank you @robertoostenveld for your suggestion, the micromed reader from Neo ([this one](https://github.com/NeuralEnsemble/python-neo/blob/master/neo/rawio/micromedrawio.py)) extracts more information from the files than the fieldtrip reader. It was actually helpful for my project to participate in this topic, I can extract the EEG date and time from the micormed files now!