mne-tools/mne-python #10526: 3D viz doesn't appear on system with older OpenGL drivers, how to fix?

hoechenberger, 2022-04-14:
So I don't think this is an actual bug at our (or, rather, pyvista's) end, but something that today prevented a user from following parts of our workshop.

The user had installed MNE-Python into a new `conda` environment following our official installation instructions.

When trying to start the coregistration UI, it crashed:

```
Error message: 2022-04-14 12:13:12.859 (   0,833s) [        61391740]vtkOpenGLRenderWindow.c:493    ERR| vtkEGLRenderWindow (0x56159d2f7f80): GLEW could not be initialized: Missing GL version
ERROR:root:GLEW could not be initialized: Missing GL version
Segmentation fault (core dumped)
```
Output of `glxinfo |grep OpenGL`:
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) UHD Graphics (CML GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 20.0.8
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 20.0.8
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 20.0.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:
```
So we have `OpenGL version string: 3.0 Mesa 20.0.8` which apparently is too old?

Would the only way to fix this be upgrading the user's graphics drivers? Or just MESA? Or…?

This is what `pyvista.GPUInfo()` produces:

```
2022-04-14 12:18:13.587 (   4.887s) [        23B16740]vtkOpenGLRenderWindow.c:499    ERR| vtkEGLRenderWindow (0x55881caf9290): Unable to find a valid OpenGL 3.2 or later implementation. Please update your video card driver to the latest version. If you are using Mesa please make sure you have version 11.2 or later and make sure your driver in Mesa supports OpenGL 3.2 such as llvmpipe or openswr. If you are on windows and using Microsoft remote desktop note that it only supports OpenGL 3.2 with nvidia quadro cards. You can use other remoting software such as nomachine to avoid this issue.
ERROR:root:Unable to find a valid OpenGL 3.2 or later implementation. Please update your video card driver to the latest version. If you are using Mesa please make sure you have version 11.2 or later and make sure your driver in Mesa supports OpenGL 3.2 such as llvmpipe or openswr. If you are on windows and using Microsoft remote desktop note that it only supports OpenGL 3.2 with nvidia quadro cards. You can use other remoting software such as nomachine to avoid this issue.
2022-04-14 12:18:13.588 (   4.888s) [        23B16740]     vtkOpenGLState.cxx:1795  WARN| Hardware does not support the number of textures defined.
2022-04-14 12:18:13.588 (   4.888s) [        23B16740]     vtkOpenGLState.cxx:1795  WARN| Hardware does not support the number of textures defined.
2022-04-14 12:18:13.590 (   4.890s) [        23B16740]     vtkOpenGLState.cxx:1795  WARN| Hardware does not support the number of textures defined.
2022-04-14 12:18:13.590 (   4.890s) [        23B16740]     vtkOpenGLState.cxx:1795  WARN| Hardware does not support the number of textures defined.
2022-04-14 12:18:13.590 (   4.890s) [        23B16740]     vtkOpenGLState.cxx:1795  WARN| Hardware does not support the number of textures defined.
2022-04-14 12:18:13.592 (   4.892s) [        23B16740]     vtkOpenGLState.cxx:1795  WARN| Hardware does not support the number of textures defined.
2022-04-14 12:18:13.592 (   4.892s) [        23B16740]     vtkOpenGLState.cxx:1795  WARN| Hardware does not support the number of textures defined.
2022-04-14 12:18:13.598 (   4.898s) [        23B16740]   vtkShaderProgram.cxx:437    ERR| vtkShaderProgram (0x55881ce6ed60): 1: #version 150
2: #ifndef GL_ES
3: #define highp
4: #define mediump
5: #define lowp
6: #endif // GL_ES
7: #define attribute in
8: #define varying out
9:
10:
11: /*=========================================================================
12:
13:   Program:   Visualization Toolkit
14:   Module:    vtkPolyDataVS.glsl
15:
16:   Copyright (c) Ken Martin, Will Schroeder, Bill Lorensen
17:   All rights reserved.
18:   See Copyright.txt or http://www.kitware.com/Copyright.htm for details.
19:
20:      This software is distributed WITHOUT ANY WARRANTY; without even
21:      the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR
22:      PURPOSE.  See the above copyright notice for more information.
23:
24: =========================================================================*/
25:
26: in vec4 vertexMC;
27:
28:
29:
30: // frag position in VC
31: out vec4 vertexVCVSOutput;
32:
33: // optional normal declaration
34: //VTK::Normal::Dec
35:
36: // extra lighting parameters
37: //VTK::Light::Dec
38:
39: // Texture coordinates
40: //VTK::TCoord::Dec
41:
42: // material property values
43: //VTK::Color::Dec
44:
45: // clipping plane vars
46: //VTK::Clip::Dec
47:
48: // camera and actor matrix values
49: uniform mat4 MCDCMatrix;
50: uniform mat4 MCVCMatrix;
51:
52: // Apple Bug
53: //VTK::PrimID::Dec
54:
55: // Value raster
56: //VTK::ValuePass::Dec
57:
58: // picking support
59: //VTK::Picking::Dec
60:
61: void main()
62: {
63:   //VTK::CustomBegin::Impl
64:
65:   //VTK::Color::Impl
66:
67:   //VTK::Normal::Impl
68:
69:   //VTK::TCoord::Impl
70:
71:   //VTK::Clip::Impl
72:
73:   //VTK::PrimID::Impl
74:
75:   vertexVCVSOutput = MCVCMatrix * vertexMC;
76:   gl_Position = MCDCMatrix * vertexMC;
77:
78:
79:   //VTK::ValuePass::Impl
80:
81:   //VTK::Light::Impl
82:
83:   //VTK::Picking::Impl
84:
85:   //VTK::CustomEnd::Impl
86: }
ERROR:root:1: #version 150
2022-04-14 12:18:13.598 (   4.899s) [        23B16740]   vtkShaderProgram.cxx:438    ERR| vtkShaderProgram (0x55881ce6ed60): 0:1(10): error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES
ERROR:root:error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES
Segmentation fault (core dumped)
```

Finally, here's `mne sys_info`:
```
Platform:         Linux-5.4.0-104-generic-x86_64-with-glibc2.27
Python:           3.9.12 | packaged by conda-forge | (main, Mar 24 2022, 23:25:59)  [GCC 10.3.0]
Executable:       /home/ar266842/anaconda3/envs/mne/bin/python3.9
CPU:              x86_64: 8 cores
Memory:           15.2 GB
mne:              1.0.1
numpy:            1.22.3 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:            1.8.0
matplotlib:       3.5.1 {backend=QtAgg}
sklearn:          1.0.2
numba:            Not found
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             1.5.0
cupy:             Not found
pandas:           1.4.2
pyvista:          0.33.3 {OpenGL could not be initialized}
pyvistaqt:        0.9.0
ipyvtklink:       Not found
vtk:              9.1.0
PyQt5:            5.9.2
ipympl:           Not found
pooch:            v1.6.0
mne_bids:         0.10
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
```

Any suggestions appreciated!

hoechenberger, 2022-04-14:
Forgot to add:

We also tried starting Coreg with `advanced_rendering=False` and we also set the environment variables:
```
"MNE_3D_OPTION_ANTIALIAS": "false",
"MNE_3D_OPTION_DEPTH_PEELING": "false",
"MNE_3D_OPTION_SMOOTH_SHADING": "false",
```

larsoner, 2022-04-14:
20.0.8 should be fine for software rendering. It looks like it's trying to use the Intel hardware graphics drivers. They were a nightmare 10-15 years ago, sounds like the situation perhaps has not improved much :(

You could try `LIBGL_ALWAYS_SOFTWARE=true` to try to force it to use Gallium/swrast/LLVMPIPE

hoechenberger, 2022-04-14:
> You could try `LIBGL_ALWAYS_SOFTWARE=true` to try to force it to use Gallium/swrast/LLVMPIPE

Thanks, will try!

PR #12042 (2023-10-01): DOC: Remove make test
Skipped CIs because it seems unnecessary based on `git grep "make test"` and `"make pytest"`

Closes #11768
Closes #10526
closed 2023-10-01 by drammock