mne-tools/mne-python #10529: export.export_raw does not write fids information in SET file

cmadjar, 2022-04-14:
#### Describe the bug

`export_raw` does not exports fids information of SET files.

#### Steps to reproduce

>>> import mne
>>> import os
>>> file = "/Users/cmadjar/Desktop/eeg_test_data/2022-04-13/003_BaDaNovel_Rutgers_SM.set"
>>> raw = mne.io.read_raw_eeglab(input_fname=file, preload=False, verbose=True)
Reading /Users/cmadjar/Documents/MATLAB/eeglab2022.0/sample_data/eeglab_data.fdt

>>> raw.get_montage()
<DigMontage | 0 extras (headshape), 0 HPIs, 3 fiducials, 129 channels>
>>> raw.get_montage().get_positions()
... 'coord_frame': 'head', 'nasion': array([ 6.34122109e-16,  1.03559999e+01, -2.69400001e+00]), 'lpa': array([-6.71199989,  0.046     , -3.71199989]), 'rpa': array([ 6.71199989,  0.046     , -3.71199989]), 'hsp': None, 'hpi': None

>>> out_file = "/Users/cmadjar/Desktop/eeg_test_data/2022-04-13/test.set"
>>> mne.export.export_raw(fname=out_file, raw=raw, fmt='eeglab')
Reading 0 ... 340244  =      0.000 ...   680.488 secs...
>>> raw_test = mne.io.read_raw_eeglab(input_fname=out_file, preload=False, verbose=True)
>>> raw_test.get_montage()
<DigMontage | 0 extras (headshape), 0 HPIs, 0 fiducials, 129 channels>       # <---------------------
>>> raw_test.get_montage().get_positions()
... 'coord_frame': 'head', 'nasion': None, 'lpa': None, 'rpa': None, 'hsp': None, 'hpi': None       # <---------------------

#### Expected results

The fids should be exported to the SET file when creating the file from raw.

#### Actual results

See lines above flagged with `      # <---------------------`

#### Additional information

Built `mne` using the following commit: https://github.com/mne-tools/mne-python/commit/706dd3a36096e88a12b3bfb207e348c26fe8625c 

```
>>> mne.sys_info()
Platform:         macOS-11.3-x86_64-i386-64bit
Python:           3.9.12 (main, Mar 26 2022, 15:52:10)  [Clang 13.0.0 (clang-1300.0.29.30)]
Executable:       /Users/cmadjar/Development/python_venv/python_intel_3.9.4/eeg2bids_py3.9.12/bin/python
CPU:              i386: 8 cores
Memory:           Unavailable (requires "psutil" package)
mne:              1.1.dev0
numpy:            1.21.5 {blas=openblas, lapack=openblas}
scipy:            1.8.0
matplotlib:       3.5.1 {backend=MacOSX}

sklearn:          1.0.2
numba:            0.55.1
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
cupy:             Not found
pandas:           1.4.1
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             Not found
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         0.10
mne_nirs:         Not found
mne_features:     0.2
mne_qt_browser:   Not found
mne_connectivity: Not found
```

agramfort, 2022-04-14:
please report this issue here https://github.com/jackz314/eeglabio/issues

cc @jackz314

closed 2022-04-14 by agramfort

cmadjar, 2022-04-14:
Thank you @agramfort. done! https://github.com/jackz314/eeglabio/issues/5