mne-tools/mne-python #10530: MNE browser does not use specified theme when using plot()

a-allard, 2022-04-14:
#### Describe the bug
When using EEG epochs the theme argument to the plot function is ignored.

#### Steps to reproduce
1. Load EEG data and break it into epochs.
2. Select epoch and plot specifying a theme other than the default theme
3. ```eps[1800].plot(theme='dark')```

(For ref:)
[image]

#### Expected results
Use the theme specified in the function call

#### Actual results
The theme set in `MNE_BROWSER_THEME` is used and the specified theme is ignored.

#### Additional information
Platform:         Linux-5.13.0-39-generic-x86_64-with-glibc2.31
Python:           3.10.4 (main, Mar 24 2022, 16:12:56) [GCC 9.4.0]
Executable:       /home/allard/pymain/bin/python3.10
CPU:              x86_64: 24 cores
Memory:           62.5 GB

mne:              1.0.1
numpy:            1.21.6 {blas=openblas, lapack=openblas}
scipy:            1.8.0
matplotlib:       3.5.1 {backend=Qt5Agg}

sklearn:          1.0.2
numba:            0.55.1
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             Not found
cupy:             Not found
pandas:           1.4.2
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
PyQt5:            5.15.6
ipympl:           Not found
pooch:            v1.6.0

mne_bids:         0.10
mne_nirs:         Not found
mne_features:     0.2
mne_qt_browser:   0.3.0
mne_connectivity: Not found

larsoner, 2022-04-14:
I think this was fixed by https://github.com/mne-tools/mne-python/pull/10500 which got merged just after 1.0.1 was released. Can you try latest MNE-Python `main` and see if it helps? You can do for example:
```
pip install --upgrade https://github.com/mne-tools/mne-python/zipball/main
```

a-allard, 2022-04-14:
@larsoner 

Installing the dev version solved it.  Sorry I didn't see #10500 when I was checking for the issue.  Closing this out.

closed 2022-04-14 by a-allard

larsoner, 2022-04-15:
Thanks for the reminder that we need to cut a 1.0.2 release!