mne-tools/mne-python #10532: SSVEP tutorial plot issue

emmaducos, 2022-04-15:
#### Describe the bug
While trying to plot the PSD vs SNR, the figure would not plot (even if it was opening and no issue reported in the terminal)
not sure why.

### steps to reproduce
doing the tutorial on SSVEP accessible here : https://mne.tools/stable/auto_tutorials/time-freq/50_ssvep.html
with mne 1.0 on pycharm, with a conda environment using mne-qt, on a windows 11 machine, using the tutorial data.

#### Expected results
tutorial script should work with the data provided by the tutorial

#### Actual results
it worked when I replaced `fig.show()` with `plt.show()` (on l. 281 of the downloaded python script)

#### Additional information
CPU:              Intel64 Family 6 Model 142 Stepping 12, GenuineIntel: 8 cores
Memory:           15.7 GB
mne:              1.0.0
numpy:            1.21.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:            1.7.3
matplotlib:       3.5.1 {backend=QtAgg}
sklearn:          1.0.2
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
cupy:             Not found
pandas:           1.4.1
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
PyQt5:            5.12.3
ipympl:           Not found
pooch:            v1.6.0
mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.2.5
mne_connectivity: Not found

agramfort, 2022-04-15:
I cannot replicate the problem on my machine in ipython

> Message ID: ***@***.***>
>

emmaducos, 2022-04-15:
this is the list of the packages used in my conda environment: 
# Name                    Version                   Build  Channel
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
atomicwrites              1.4.0                    pypi_0    pypi
attrs                     21.4.0                   pypi_0    pypi
blas                      1.0                         mkl
bottleneck                1.3.4           py310h9128911_0
brotli                    1.0.9                h8ffe710_6    conda-forge
brotli-bin                1.0.9                h8ffe710_6    conda-forge
brotlipy                  0.7.0           py310he2412df_1003    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2022.3.29            haa95532_0
certifi                   2021.5.30       py310haa95532_0
cffi                      1.15.0          py310hcbf9ad4_0    conda-forge
charset-normalizer        2.0.12             pyhd8ed1ab_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
colorspacious             1.1.2              pyh24bf2e0_0    conda-forge
cryptography              36.0.2          py310ha857299_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
daal4py                   2021.4.0        py310h2fba4dc_0
dal                       2021.4.0           haa95532_729
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
fonttools                 4.31.2          py310he2412df_0    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
icc_rt                    2019.0.0             h0cc432a_1
icu                       69.1                 h0e60522_0    conda-forge
idna                      3.3                pyhd8ed1ab_0    conda-forge
iniconfig                 1.1.1                    pypi_0    pypi
intel-openmp              2021.4.0          haa95532_3556
jbig                      2.1               h8d14728_2003    conda-forge
jinja2                    3.1.1              pyhd8ed1ab_0    conda-forge
joblib                    1.1.0              pyhd8ed1ab_0    conda-forge
jpeg                      9e                   h8ffe710_0    conda-forge
kiwisolver                1.4.0           py310h476a331_0    conda-forge
lcms2                     2.12                 h2a16943_0    conda-forge
lerc                      3.0                  h0e60522_0    conda-forge
libbrotlicommon           1.0.9                h8ffe710_6    conda-forge
libbrotlidec              1.0.9                h8ffe710_6    conda-forge
libbrotlienc              1.0.9                h8ffe710_6    conda-forge
libclang                  13.0.1          default_h81446c8_0    conda-forge
libdeflate                1.10                 h8ffe710_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
libtiff                   4.3.0                hc4061b1_3    conda-forge
libwebp                   1.2.2                h57928b3_0    conda-forge
libwebp-base              1.2.2                h8ffe710_1    conda-forge
libxcb                    1.13              hcd874cb_1004    conda-forge
libzlib                   1.2.11            h8ffe710_1014    conda-forge
lz4-c                     1.9.3                h8ffe710_1    conda-forge
m2w64-gcc-libgfortran     5.3.0                         6    conda-forge
m2w64-gcc-libs            5.3.0                         7    conda-forge
m2w64-gcc-libs-core       5.3.0                         7    conda-forge
m2w64-gmp                 6.1.0                         2    conda-forge
m2w64-libwinpthread-git   5.0.0.4634.697f757               2    conda-forge
markupsafe                2.1.1           py310he2412df_0    conda-forge
matplotlib                3.5.1           py310h5588dad_0    conda-forge
matplotlib-base           3.5.1           py310h79a7439_0    conda-forge
mkl                       2021.4.0           haa95532_640
mkl-service               2.4.0           py310h2bbff1b_0
mkl_fft                   1.3.1           py310ha0764ea_0
mkl_random                1.2.2           py310h4ed8f06_0
mne-base                  1.0.0              pyhd8ed1ab_1    conda-forge
mne-qt-browser            0.2.5           py310h5588dad_1    conda-forge
msys2-conda-epoch         20160418                      1    conda-forge
munkres                   1.1.4              pyh9f0ad1d_0    conda-forge
numexpr                   2.8.1           py310hb57aa6b_1
numpy                     1.21.5          py310h4c31df0_0
numpy-base                1.21.5          py310hedd7904_0
openjpeg                  2.4.0                hb211442_1    conda-forge
openssl                   1.1.1n               h2bbff1b_0
[... 34 lines cut ...]
tbb                       2021.5.0             h2d74725_0    conda-forge
threadpoolctl             3.1.0              pyh8a188c0_0    conda-forge
tk                        8.6.12               h8ffe710_0    conda-forge
tomli                     2.0.1                    pypi_0    pypi
tornado                   6.1             py310he2412df_2    conda-forge
tqdm                      4.63.1             pyhd8ed1ab_0    conda-forge
tzdata                    2022a                h191b570_0    conda-forge
ucrt                      10.0.20348.0         h57928b3_0    conda-forge
[... cut ...]
xz                        5.2.5                h62dcd97_1    conda-forge
zlib                      1.2.11            h8ffe710_1014    conda-forge
zstd                      1.5.2                h6255e5f_0    conda-forge

maybe some of them are interfering together ? 
I tried to fix the backend of matplotlib manually but it did not change anything.

Since I solved my issue, this thread is just a warning that I had issues, but this is not important per se. 
Thanks for answering this quickly. have a nice weekend