mne-tools/mne-python #10533: add convenient function to "prepare" emptyroom raw file for maxwell_filter

agramfort, 2022-04-15:
this is the type of code that we currently need to write to process correctly emtpyroom file the same way as a subject raw file.

```python
raw_er = mne.io.read_raw_fif('.../emptyroom.fif', allow_maxshield="yes")
raw_er.info["bads"] = bads
raw_er.info["dev_head_t"] = raw.info["dev_head_t"]
raw_er.set_montage(raw.get_montage())
raw_er = mne.preprocessing.maxwell_filter(
    raw_er, origin="auto", calibration=cal_fname, cross_talk=ct_fname,
)
```

the proposal is to simplify this to:

```python
raw_er = mne.io.read_raw_fif('.../emptyroom.fif', allow_maxshield="yes")
raw_er = mne.preprocessing.prepare_maxwell_filter_emptyroom(raw_er, raw)
raw_er = mne.preprocessing.maxwell_filter(
    raw_er, origin="auto", calibration=cal_fname, cross_talk=ct_fname,
)
```

or maybe

```
raw_er = mne.preprocessing.maxwell_filter_prepare_emptyroom(raw_er, raw)
```

discussed with @larsoner and @hoechenberger

hoechenberger, 2022-04-16:
How do we deal with existing `bads` in the empty-room? If the raw_er contains bads that are not present in the raw, raw_er should probably get the union of `raw.info['bads']` and `raw_er.info['bads']`. But then we'd again have a mismatch between raw and raw_er, unless we modify `raw.info['bads']` too (also put the union of bads there).

larsoner, 2022-04-18:
> But then we'd again have a mismatch between raw and raw_er, unless we modify raw.info['bads'] too (also put the union of bads there).

I would warn if the `raw_erm` has bads that `raw` does not

hoechenberger, 2022-04-18:
> I would warn if the `raw_erm` has bads that `raw` does not

In the PR (#10537) I've added a parameter to the function, allowing the user to explicitly specify what should happen to the bads
Doing "nothing" and warning  is currently not implemented, but I could add this!

PR #10537 (2022-04-21): MRG: Add convenience function to "prepare" emptyroom raw file for maxwell_filter
Fixes #10533

This is a first quick draft, I'm looking for feedback.

cc @agramfort
closed 2022-04-21 by hoechenberger