mne-tools/mne-python #10539: Plot raw side-by-side

mileshenrichs, 2022-04-17:
#### Describe the new feature or enhancement
Enable inline plotting of multiple raw plots side by side. Two different raw plots can be compared now by using qt5 interactive plot mode and placing plot windows side by side, but it would be nice to be able to do this inline.

#### Describe your proposed implementation
Allow a figure or axes argument to be passed into the `raw.plot()` method so that subplots can be configured prior to plotting, then the plot can be drawn onto the subplots which can be arranged side-by-side.

If there's already a way to do this please let me know! I haven't been able to find it during my searches through documentation thus far. Thank you!

agramfort, 2022-04-17:
you need this for interactive exploration or saving figures?

> Message ID: ***@***.***>
>

larsoner, 2023-10-01:
Closing due to non-response. Also one potential workaround would be to create two mne-qt-browser instances and use Qt methods to place the windows wherever you want on your screen using standard geometry commands. The same might also be possible in matplotlib

closed 2023-10-01 by larsoner