mne-tools/mne-python #10551: Does the coreg GUI work with Python 3.10?

cbrnr, 2022-04-20:
I might be wrong, but it looks like `mne coreg` requires Python < 3.10 to work (simply because `vtk` does not have binary wheels for Python 3.10). If that's the case, should we add a note somewhere in the documentation (maybe in the paragraph describing the installation with 3D visualization [here](https://mne.tools/stable/install/manual_install.html))?

larsoner, 2022-04-20:
There are VTK 3.10 wheels on conda-forge (our recommended installation method). Wheels should show up on PyPi around May 6 because 9.2 is 10-compatible https://discourse.vtk.org/t/vtk-9-2-0-release-cycle/8149

cbrnr, 2022-04-20:
I guess I'll have to wait until then.

larsoner, 2022-04-20:
In the meantime you can try https://github.com/pyvista/pyvista-wheels if you want

cbrnr, 2022-04-20:
Nice, thanks @larsoner! I will try that!

cbrnr, 2022-04-20:
```
pip install ~/Downloads/vtk-9.1.0.dev0-cp310-cp310-macosx_12_0_x86_64.whl
ERROR: vtk-9.1.0.dev0-cp310-cp310-macosx_12_0_x86_64.whl is not a supported wheel on this platform.
```

👀 – my MacBook doesn't support Monterey anymore 😢.