mne-tools/mne-python #10553: Dropdown menu on website is using the wrong theme

hoechenberger, 2022-04-20:
Light mode:
[image]

Dark mode:
[image]

Also I have this behavior with the toggle button, which is clearly a bug:

https://user-images.githubusercontent.com/2046265/164280586-ee0cfd69-b7c9-4a92-9fc3-39e194fcbf43.mov

This is with Safari 15.4 on macOS.

cc @larsoner

larsoner, 2022-04-20:
What's the problem with the drop-down? The colors are different and they should be depending on light/dark mode

hoechenberger, 2022-04-20:
We used to have a blue dropdown in the Light Theme before and that's what I'd expect when using a light theme – the Bootstrap "primary" buttons are supposed to be blue-ish. Therefore I thought it's a bug.

If it's not, then I propose to revert the color change to the light-themed button, please :)

hoechenberger, 2022-04-20:
But why did the color suddenly change? This button used to be blue until recently…

larsoner, 2022-04-20:
Not sure, you could trace back in the `mne-tools.github.io` repo if you're really curious. But personally I would just look into fixing it by removing our customizations where possible...

larsoner, 2022-05-06:
I think this is an intentional change at the pydata-sphinx-theme end as it now uses `var(--pst-color-active-navigation)` via `#version_switcher_button`. I think we should close this as I'd rather not deviate from their theme on this

hoechenberger, 2022-05-06:
Thanks for investigating this!

closed 2022-05-06 by hoechenberger