mne-tools/mne-python #10562: If we are missing an optional module or package to operate, don't just complain about it being missing, but also give advice on how to install it

hoechenberger, 2022-04-26:
For example, pymatreader and eeglabio are optimal and simply telling users we need the module for certain things apparently doesn't prove to be too helpful, see e.g.

https://mne.discourse.group/t/error-the-pymatreader-module-is-needed-but-it-could-not-be-imported/4783/3?u=richard

cc @cbrnr

cbrnr, 2022-04-26:
Makes sense. Something like "Use pip or conda to install it."? Or more specific?

hoechenberger, 2022-04-26:
I would print the exact command to be used :)

hoechenberger, 2022-04-26:
Yes, this is how I had imagined it!

cbrnr, 2022-04-26:
PyPI package names are case-insensitive, so we could user all lowercase for both conda and pip package names.

larsoner, 2022-04-26:
> PyPI package names are case-insensitive, so we could user all lowercase for both conda and pip package names.

Indeed but I think it's nice to match the case if we can. It's easy enough to do...

From a [previous conversation](https://github.com/mne-tools/mne-bids/issues/1002#issuecomment-1108661727) I think @scott-huberty plans to work on this -- @scott-huberty feel free to see above for the suggested wording

cbrnr, 2022-04-26:
About that package... we also discussed that we might want to replace `edflib-python` with `pyedflib` because the latter is much much faster.

larsoner, 2022-04-26:
Yep that's https://github.com/mne-tools/mne-python/issues/9883 (and let's keep that conversation over there)

scott-huberty, 2022-04-26:
> > PyPI package names are case-insensitive, so we could use all lowercase for both conda and pip package names.
> 
> Indeed but I think it's nice to match the case if we can. It's easy enough to do...
> 
> From a [previous conversation](https://github.com/mne-tools/mne-bids/issues/1002#issuecomment-1108661727) I think @scott-huberty plans to work on this -- @scott-huberty feel free to see above for the suggested wording

Sounds good, I just have to finish a paper revision first but then will start working on this. I think the suggestions above are helpful. so as I understand it, the new-and-improved error message should address the following: 
1. provide more verbose instructions on how to install, per the suggestions above.
2. [(per this conversation)](https://github.com/mne-tools/mne-bids/issues/1002#issuecomment-1108752964), refer to the package installation name instead of the import name for packages where these are different. To this end it would be helpful to know if there are any other optional dependencies in MNE where the installation name and import namespace differ (right now I am aware of two):

-    EDFlib: "EDFlib-Python"
-   sklearn : "scikit-learn"

larsoner, 2022-04-26:
Sounds good to me!

scott-huberty, 2022-05-25:
Sorry for the delay here.. I've submitted a PR - I made the aforementioned revisions to `_soft_import` in `mne.utils.check`, to give more verbose instructions to pip install missing dependencies, and to make sure the correct pip_name is suggested, for the set of MNE optional dependencies that have differing names for pip installing and importing.

hoechenberger, 2022-06-15:
Very nice!