mne-tools/mne-python #10564: Unsupported array dtype in compute forward

maedoc, 2022-04-26:
#### Describe the bug
When computing a forward solution on HCP MEG data (following the [tutorial](https://mne.tools/mne-hcp/auto_tutorials/plot_compute_forward.html#sphx-glr-auto-tutorials-plot-compute-forward-py)), Numba complains about a dtype.

#### Steps to reproduce
I'm copying and pasting code from the MNE-HCP tutorial here

https://mne.tools/mne-hcp/auto_tutorials/plot_compute_forward.html#sphx-glr-auto-tutorials-plot-compute-forward-py

#### Expected results
I expected the `mne.make_forward_solution` call to succeed.

#### Actual results

```
>>> fwd = mne.make_forward_solution(info, trans=head_mri_t, bem=bem_sol, src=src_subject)
...
TypingError: Failed in nopython mode pipeline (step: nopython frontend)
non-precise type pyobject
During: typing of argument at /Users/duke/miniforge3/lib/python3.9/site-packages/mne/forward/_compute_forward.py (132)

File "../../miniforge3/lib/python3.9/site-packages/mne/forward/_compute_forward.py", line 132:
def _do_lin_field_coeff(bem_rr, tris, tn, ta, rmags, cosmags, ws, bins):
    <source elided>
    """
    coeff = np.zeros((bins[-1] + 1, len(bem_rr)))
    ^ 

This error may have been caused by the following argument(s):
- argument 1: Unsupported array dtype: >i4
```

I guess this may be an m1 arm64 issue about integer sizes but not sure.  This is Numba running natively and not through Rosetta 2 x86_64 emulation.

#### Additional information
```
Platform:         macOS-12.3.1-arm64-arm-64bit
Python:           3.9.7 | packaged by conda-forge | (default, Sep 29 2021, 19:24:02)  [Clang 11.1.0 ]
Executable:       /Users/duke/miniforge3/bin/python3
CPU:              arm: 10 cores
Memory:           Unavailable (requires "psutil" package)
mne:              1.0.2
numpy:            1.20.3 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:            1.7.3
matplotlib:       3.5.0 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.0.1
numba:            0.54.1
nibabel:          3.2.2
nilearn:          Not found
dipy:             Not found
cupy:             Not found
pandas:           1.3.4
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
PyQt5:            Not found
ipympl:           Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
```

larsoner, 2022-04-26:
> Does your jit wrapper default to nopython mode?

Yes and this is intentional. I think we should make sure that the `bins[-1]` is a type Numba can handle. It's not immediately clear to me why it wouldn't be, but it should be easy enough to fix.

If you put the `@jit` decorator back in place, but one stack level up you pass `bins.astype(int)` or `bins.astype(np.int64)` does it work? If so, this is effectively what we want to accomplish, and we need to figure out why `bins` is not one of these types for the HCP data...

maedoc, 2022-04-26:
`bins.dtype` is `int64`, and passing `bins.astype('i')` doesn't work either. The relevant code up stack calls this with a genexpr which doesn't work in the debugger.

maedoc, 2022-04-26:
ah, the error-message explains why: `>i4` is a big-endian dtype.

larsoner, 2022-04-26:
Oh I can't count/read, it's not the `bins` it's the `tris` that have `>i4` (this is arg 1). If you `.astype(int)` or `.astype(np.int64)` this does it work?

maedoc, 2022-04-26:
yes, `surf['tris'].astype('i')` does the trick.