mne-tools/mne-python #10566: [ENH] Auto-document parameters with non-`None` default values

alexrockhill, 2022-04-26:
I find myself writing "The default is x" in a lot of parameter docstrings. What do people think about checking if the parameter is not `None` and, if so, appending that verbiage automatically. One major pro is that it's not repetitive in case the default parameter value changes so it avoids a potential mismatch with the docstring.

alexrockhill, 2022-04-26:
There are some cases where units are in the default but I think something like

```
duration : float
    The time in seconds to make each epoch. The default is 3.
```

Would be okay although I could see something like 

```
duration : float
    The time in seconds to make each epoch. The default is 3 seconds.
```

being documented currently and that might be preferable.

agramfort, 2022-04-26:
do you know any project doing this?

> Message ID: ***@***.***>
>

alexrockhill, 2022-04-26:
Hmm, I don't. It might be apparent from the documentation but since I'm not that wide of a contributor I probably wouldn't be the best to ask. Maybe it's not necessary but I do appreciate when you print the `__doc__` if the default parameters are listed.

cbrnr, 2022-04-26:
I would not mention the defaults in the docstring at all. It is already evident from the function header anyway.

alexrockhill, 2022-04-26:
@cbrnr, that would be ideal but in a lot of our documentation we `print(__doc__)` this seems like a super basic question but how do you know what the default parameters of a function are from its docstring? That seems pretty essential in using that function. https://stackoverflow.com/questions/218616/how-to-get-method-parameter-names seems to be not a great solution, but maybe there is some python 3 stuff I don't know about.

cbrnr, 2022-04-26:
You cannot tell from the docstring if you don't explicitly include the defaults, but our website shows the function signatures including default arguments. Isn't this sufficient? Why do you need to have them in the docstring as well?

alexrockhill, 2022-04-26:
What if you don't have access to internet? Also I think a lot of people work very comfortably in a python console and it's rather convenient to access documentation there.

cbrnr, 2022-04-26:
The interactive help in IPython does show the signature, e.g.

```
Signature:
mne.channels.combine_channels(
    inst,
    groups,
    method='mean',
    keep_stim=False,
    drop_bad=False,
)
Docstring:
Combine channels based on specified channel grouping.
...
```

Same for the regular interpreter:

```
Help on function combine_channels in module mne.channels.channels:

combine_channels(inst, groups, method='mean', keep_stim=False, drop_bad=False)
    Combine channels based on specified channel grouping.
...
```

closed 2022-04-26 by alexrockhill

cbrnr, 2022-04-26:
Yes, I think so too. I would remove all mentions of defaults from our docstrings. This also avoids situations where we change the default and forget to update the docstring.