mne-tools/mne-python #10570: mne.fit_dipole throwing AssertionError: (blank) when n_jobs=-1

mjcourte, 2022-04-26:
#### Describe the bug
When setting `n_jobs=-1`  as an argument of `mne.fit_dipole()`, a blank AssertionError is thrown.
Reading up the trace, one can see there is an `assert n_jobs > 0`  within `mne/surface.py`.
Elsewhere in MNE-Py, setting `n_jobs=-1` is valid and selects maximum available workers, and the docs for `fit_dipole` state this.
Setting `n_jobs=16` runs successfully, where 16 is the maximum workers on my system, and the number of workers when using -1 in the steps prior to dipole fitting.

#### Steps to reproduce
Run:
```
dip = mne.fit_dipole(..., n_jobs=-1)
```

#### Expected results

The dipole fit to complete using maximum workers.

#### Actual results
```
(snip)
~\anaconda3\envs\mne\lib\site-packages\mne\surface.py in _points_outside_surface(rr, surf, n_jobs, verbose)
    561     rr = np.atleast_2d(rr)
    562     assert rr.shape[1] == 3
--> 563     assert n_jobs > 0
    564     parallel, p_fun, _ = parallel_func(_get_solids, n_jobs)
    565     tot_angles = parallel(p_fun(surf['rr'][tris], rr)

AssertionError: 
```

#### Additional information
<details>
  <summary>Output from `mne.sys_info()`</summary>
  
  Platform:       Windows-10-10.0.22000-SP0
  Python:         3.9.9 | packaged by conda-forge | (main, Dec 20 2021, 02:36:06) [MSC v.1929 64 bit (AMD64)]
  Executable:     C:\Users\Matt\anaconda3\envs\mne\python.exe
  CPU:            Intel64 Family 6 Model 167 Stepping 1, GenuineIntel: 16 cores
  Memory:         63.8 GB
  
  mne:            0.24.1
  numpy:          1.22.0 {blas=NO_ATLAS_INFO, lapack=lapack}
  scipy:          1.7.3
  matplotlib:     3.5.1 {backend=module://matplotlib_inline.backend_inline}
  
  sklearn:        1.0.2
  numba:          0.53.1
  nibabel:        3.2.1
  nilearn:        0.8.1
  dipy:           1.4.1
  cupy:           Not found
  pandas:         1.3.5
  mayavi:         4.7.2
  pyvista:        0.33.2 {OpenGL 4.5.0 NVIDIA 496.49 via NVIDIA GeForce GTX 1070/PCIe/SSE2}
  pyvistaqt:      0.6.0
  ipyvtklink:     Not found
  vtk:            9.0.3
  PyQt5:          5.12.3
  ipympl:         Not found
  mne_qt_browser: 0.1.7
  pooch:          v1.5.2
  C:\Users\Matt\anaconda3\envs\mne\lib\site-packages\mne\utils\check.py:728: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
    if LooseVersion(version) >= LooseVersion('5.10'):
  C:\Users\Matt\anaconda3\envs\mne\lib\site-packages\mne\utils\check.py:728: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
    if LooseVersion(version) >= LooseVersion('5.10'):
</details>

larsoner, 2022-04-27:
Can you see if https://github.com/mne-tools/mne-python/pull/10567 fixes it?

mjcourte, 2022-04-27:
Tested to your n_jobs branch mne 1.1.dev0. fits_dipole now working as intended with -1 jobs.

Thanks!

PR #10567 (2022-04-29): ENH: Add n_jobs=None support
Closes #10522
Closes #10570

- [x] Search-and-replace `n_jobs=1` -> `n_jobs=None`
- [x] `flake8` fixes, including some insertions of `*` before verbose when noticed/easy (not trying to be complete, incremental progress good enough, don't want to make judgment calls about whether to move it farther forward than just before `verbose`)
- [x] Look through diff to make sure all new `n_jobs=None` actually make sense
- [x] Figure out how to handle our `if n_jobs == 1` conditionals that now will fail (maybe with `check_n_jobs`?)
- [x] Actually make `n_jobs=None` do more th [... cut]
closed 2022-04-29 by agramfort