mne-tools/mne-python #10577: Old mne version with installation from conda-forge on linux-64

marsipu, 2022-04-28:
#### Describe the bug
The installation of mne via conda commands on Linux doesn't install the newest version.

#### Steps to reproduce
```
conda create --name mne --channel conda-forge mne
```

#### Expected result
currently `mne==1.0.2`

#### Actual result
`mne==0.23.4`

#### mne sys_info
```
Platform:      Linux-5.13.0-40-generic-x86_64-with-glibc2.31
Python:        3.9.12 | packaged by conda-forge | (main, Mar 24 2022, 23:25:59)  [GCC 10.3.0]
Executable:    <home>/miniconda3/envs/mne/bin/python
CPU:           x86_64: 32 cores
Memory:        31.3 GB

mne:           0.23.4
numpy:         1.22.3 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.8.0
matplotlib:    3.5.1 {backend=QtAgg}

sklearn:       1.0.2
numba:         Not found
nibabel:       3.2.2
nilearn:       0.9.1
dipy:          1.5.0
cupy:          Not found
pandas:        1.4.2
mayavi:        4.7.4
pyvista:       0.34.0 {pyvistaqt=0.9.0, OpenGL 4.6.0 NVIDIA 470.103.01 via Quadro K5200/PCIe/SSE2}
vtk:           
PyQt5:         5.9.2

```

marsipu, 2022-04-28:
It seems as if mne is installed from `linux-64` and not `noarch`.
[image]

But when I try `conda create --n mne -c conda-forge/noarch mne`, the version is even lower (`mne==0.16.2`)

larsoner, 2022-04-28:
That is very strange, is your `conda` up to date?

marsipu, 2022-04-28:
Yes, `conda==4.12.0`.
I also tried `conda clean --all`, but that didn't help.

hoechenberger, 2022-04-29:
This was reported on the forum too and I'm planning to look into it this weekend.

https://mne.discourse.group/t/mne-1-0-in-conda-forge/4743

hoechenberger, 2022-04-29:
@marsipu A temporary workaround is to explicitly specify the package version, as in

```
conda create --name mne --channel conda-forge mne==1.0.2
```

hoechenberger, 2022-04-29:
@marsipu Could you please try to create the environment with `mamba` instead of `conda` and see if you run into the same issue?

hoechenberger, 2022-04-29:
@marsipu Could you please share with me the output of

```
conda info
```
and
```
conda config --show channel_priority
```

Thanks!

hoechenberger, 2022-04-30:
@larsoner Could you try the following to see if you'll run into the same issue?

```shell
mamba create -n mne-installation-test
mamba activate mne-installation-test
echo "channel_priority: flexible\nchannels:\n  - main" > $CONDA_PREFIX/.condarc
mamba create --name mne-test --channel conda-forge mne
```
The idea is to create a situation that resembles the default behavior of `conda` on systems where `Miniconda` or `Anaconda` was installed – I've been using `Miniforge` and Mambaforge` for years now so things behave differently there.

My guess is that the `channel_priority` is the problem. By default, it's `flexible`, and if the user didn't use `Miniconda` or `Miniforge`, the `main` channels are part of the "package source mix", and there might be situations where the solver finds it easier to pull in certain dependencies from `main` than from `conda-forge`, with odd consequences.

The `conda-forge` docs [explicitly mention](https://conda-forge.org/#about) that one should set `channel_priority` to `strict`, probably to avoid exactly these issues.

marsipu, 2022-04-30:
## conda info
```
     active environment : base
    active env location : /home/martins/miniconda3
            shell level : 1
       user config file : /home/martins/.condarc
 populated config files : 
          conda version : 4.12.0
    conda-build version : not installed
         python version : 3.7.7.final.0
       virtual packages : __cuda=11.4=0
                          __linux=5.13.0=0
                          __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/martins/miniconda3  (writable)
      conda av data dir : /home/martins/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/martins/miniconda3/pkgs
                          /home/martins/.conda/pkgs
       envs directories : /home/martins/miniconda3/envs
                          /home/martins/.conda/envs
               platform : linux-64
             user-agent : conda/4.12.0 requests/2.27.1 CPython/3.7.7 Linux/5.13.0-40-generic ubuntu/20.04.4 glibc/2.31
                UID:GID : 1001:1001
             netrc file : None
           offline mode : False

```

marsipu, 2022-04-30:
## conda config --show channel_priority
channel-priority: flexible

hoechenberger, 2022-04-30:
@marsipu Can you run:

```
conda config --set channel_priority strict
```

and try again to create the MNE environment?

marsipu, 2022-04-30:
Yes, this seems to solve the problem.
But it seems that the [default value](https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-channels.html#strict-channel-priority) for `channel_priority` in conda is `flexible`. Should we warn about this in the docs?
<details>
<summary>mne sys_info</summary>
Platform:         Linux-5.13.0-40-generic-x86_64-with-glibc2.31
Python:           3.10.4 | packaged by conda-forge | (main, Mar 24 2022, 17:38:57) [GCC 10.3.0]
Executable:       <home>/miniconda3/envs/mne/bin/python3.10
CPU:              x86_64: 32 cores
Memory:           31.3 GB

mne:              1.0.2
numpy:            1.21.6 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:            1.8.0
matplotlib:       3.5.1 {backend=QtAgg}

sklearn:          1.0.2
numba:            0.55.1
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             1.5.0
cupy:             Not found
pandas:           1.4.2
pyvista:          0.34.0 {OpenGL 4.5.0 NVIDIA 470.103.01 via Quadro K5200/PCIe/SSE2}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.2
vtk:              9.1.0
PyQt5:            5.12.3
ipympl:           Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.3.0
mne_connectivity: Not found
</details>

hoechenberger, 2022-04-30:
Ok, cool, thanks for checking!

The default will change to strict in conda 5.0, but I'm not sure when this will be released. So until then, we should amend the docs, yes.

PR #10580 (2022-05-02): MRG: Update installation instructions to ensure users end up with latest version of the conda-forge package
Also re-ordered the instructions, as I've seen several users end up with only 2D  support & minimal installs as these instructions were at the very top. I moved  those minimal installs down, as this is a special case that only select users will want.

 Also we're now suggesting users use <strike>mamba</strike> libmamba for the "full" install  (we keep recommending conda for the minimal installs).

Fixes #10577

cc @marsipu @larsoner @agramfort
commit 820e3ccc (2022-05-02): MRG: Update installation instructions to ensure users end up with latest version of the conda-forge package (#10580)

* Update installation instructions to ensure users end up with latest version of the conda-forge package

Also re-ordered the instructions, as I've seen several users end up with only 2D
 support & minimal installs as these instructions were at the very top. I moved
 those minimal installs down, as this is a special case that only select users will want.

 Also we're now suggesting users use mamba for the "full" install
 (we keep recommending conda for the minimal installs).

F [... cut]
closed 2022-05-02 by hoechenberger