mne-tools/mne-python #10585: [BUG] fNIRS nirx montage roundtrip failure

alexrockhill, 2022-05-02:
In running the following minimally reproducible example

```
import os.path as op
import mne
from mne.datasets.testing import data_path

testing_path = data_path(download=False)
fname_nirx_15_0 = op.join(
    testing_path, 'NIRx', 'nirscout', 'nirx_15_0_recording')

raw = mne.io.read_raw_nirx(fname_nirx_15_0)
montage = raw.get_montage()
raw.set_montage(montage)
```

I get the following stacktrace

```
ValueError                                Traceback (most recent call last)
<ipython-input-3-6ef838ccbf6c> in <module>
      9 raw = mne.io.read_raw_nirx(fname_nirx_15_0)
     10 montage = raw.get_montage()
---> 11 raw.set_montage(montage)

<decorator-gen-28> in set_montage(self, montage, match_case, match_alias, on_missing, verbose)

~/projects/mne-python/mne/io/meas_info.py in set_montage(self, montage, match_case, match_alias, on_missing, verbose)
    221         from ..channels.montage import _set_montage
    222         info = self if isinstance(self, Info) else self.info
--> 223         _set_montage(info, montage, match_case, match_alias, on_missing)
    224         return self
    225 

~/projects/mne-python/mne/channels/montage.py in _set_montage(***failed resolving arguments***)
   1094     fnirs_picks = _picks_to_idx(info, 'fnirs', allow_empty=True)
   1095     if len(fnirs_picks) > 0:
-> 1096         _set_montage_fnirs(info, mnt_head)
   1097 
   1098 

~/projects/mne-python/mne/channels/montage.py in _set_montage_fnirs(info, montage)
    875         ch = info['chs'][ch_idx]['ch_name']
    876         source, detector = ch.split(' ')[0].split('_')
--> 877         source_pos = montage.dig[montage.ch_names.index(source)
    878                                  + num_ficiduals]['r']
    879         detector_pos = montage.dig[montage.ch_names.index(detector)

ValueError: 'S1' is not in list
```

I don't know a whole lot about setting montages in fNIRS but this seems like an issue if you wanted to adjust coordinates let's say. If there's a reason fNIRS should not be able to go round-trip with a montage, perhaps a more informative error should be given as well. cc @rob-luke

alexrockhill, 2022-05-02:
Related to https://mne.discourse.group/t/topographic-visualisation-is-not-well-centred/4603/15 as well

rob-luke, 2022-05-02:
Adding get montage support for fNIRS was the next step to #10555. Are you able to open a Pr for this @alexrockhill ?

alexrockhill, 2022-05-02:
I'm not sure what the solution is here. Raise an error on `get_montage` for `fNIRS` data? That really should happen now since it's not supported. I don't have experience working with fNIRS so for this device-dependent stuff, it would probably be better to have someone with more experience. I think the main point is that there are users, running into unclear errors and posting them on the discussion board and I think a bit of work on the code as far as throwing informative errors would go a long way toward helping users and MNE developers with a more general knowledge of coding and MNE specifically help so that you don't have to answer every fNIRS question @rob-luke.

rob-luke, 2022-05-03:
There are countless places in the code base that don't work with fNIRS. I am slowly making my way through them over the last year. So if you spot any (like this) it's great if you highlight them in an issue and tag it with `fNIRS`, and open a pr to throw a better warning until it's fixed. Feel free to @ me for a review on any warning PRs incase there is an alternative way that works.

larsoner, 2022-05-03:
> I'm not sure what the solution is here. Raise an error on get_montage for fNIRS data? That really should happen now since it's not supported. I don't have experience working with fNIRS so for this device-dependent stuff,

This is actually not device dependent I don't think. S1 should be in any (sane) fNIRS montage.

I think the issue is that `set_montage` rightly expects `S1` and `D1` etc., but `get_montage` errantly returns something else (probably things like `S1_D1 760`. So the fix is to add this round-trip get/set montage at least to the fNIRS tests (TDD!) then change `get_montage` for fNIRS devices to properly return S1 and D1 etc. locations (what end up in `loc[3:6]` and `loc[6:9]`), not their midpoints at S1_D1 (what ends up in `loc[:3]`) which is what I assume get returned now.

rob-luke, 2022-05-04:
I was planning to tackle get montage in the next week or so. But if anyone else wants to give it a try I'll hold off.

PR #10611 (2022-05-10): Add get_montage support for fNIRS
#### Reference issue
Follow up to #10555
Closes #10585

#### What does this implement/fix?
Fixes `get_montage` for fNIRS data so that a user can run `raw.set_montage(raw.get_montage())`.
fNIRS data has multiple channels between each optode (source, detector). This was already handled with `set_montage` in #9141
closed 2022-05-10 by larsoner