mne-tools/mne-python #10591: index bug when replacing report elements

aseeland, 2022-05-03:
Hello MNE-team,

I found a bug in these lines of code, that are executed when a content element gets replaced:

https://github.com/mne-tools/mne-python/blob/cc44af23232542e5a14d86922e10f6d0ea22bc1e/mne/report/report.py#L1855-L1858

The content-element list (self._content) is iterated in reverse order, but the idx from enumerate isn't. This results in the wrong element being replaced.
There are several possibilities how to fix it, and I'm not sure, what you like:
- do not reverse self._content
- when replacing calculate something like len(self._content)-idx-1
- include enumerate in reversing: for idx, content_element in reversed(list(enumerate(self._content)))

I use the current stable version (1.0.2), but the bug is also on main.

With kind regards,
Anett

hoechenberger, 2022-05-04:
Hello @aseeland, thanks for reporting the issue!

Would you be willing to try to contribute a bug fix? I can give you a hand if you don't really know where to start!

Best wishes,
Richard

larsoner, 2022-12-08:
This has been fixed in `main` and will be part of 1.3

closed 2022-12-08 by larsoner