mne-tools/mne-python #10594: CircleCI builds of documentation are once again not getting deployed?

hoechenberger, 2022-05-04:
For example, a change I made to the stable docs yesterday still doesn't show up on the website; the respective Circle build doesn't seem to have … been building _anything_..?

https://app.circleci.com/pipelines/github/mne-tools/mne-python/14066/workflows/cf92632a-1e47-49be-82fb-ee9496048932

cc @larsoner

larsoner, 2022-05-05:
(and I just did it, so docs should be updated in a couple of hours)

hoechenberger, 2022-05-05:
Thanks @larsoner

hoechenberger, 2022-05-05:
@larsoner It seems the `main` docs are also not being deployed properly :(

larsoner, 2022-05-05:
If you look at the overnight build log you should see it was a spurious download failure

hoechenberger, 2022-05-05:
Oh so we don't build & deploy upon merging into `main` anymore? I think I recall this discussion when we introduced that change … ok, thanks for letting me know!

hoechenberger, 2022-06-02:
Ah, right, I forgot about this! Thanks!