mne-tools/mne-python #10602: [ENH] visualize "adjacency" for cluster-based permutation tests

sappelhoff, 2022-05-05:
FieldTrip has "neighbor plots", see https://www.fieldtriptoolbox.org/example/neighbours/ and this screenshot:

[image]

I think this would be nice to have for the "adjacency" that we have as a parameter for MNE-Python's cluster permutation tests. Right now I trust [`mne.channels.find_ch_adjacency`](https://mne.tools/stable/generated/mne.channels.find_ch_adjacency.html?highlight=find_ch_adjacency#mne.channels.find_ch_adjacency) to do the right job, ... but it'd be super nice to:

1. check the results via something like a "neighbor plot"
2. maybe even "tweak" what's considered a neighbor (i.e., manipulate a distance threshold)

I have seen one visualization of "adjacency" [here](https://mne.tools/stable/auto_tutorials/stats-sensor-space/75_cluster_ftest_spatiotemporal.html#find-the-fieldtrip-neighbor-definition-to-setup-sensor-adjacency): 

[image]

But it's not very intuitive to use.

What do you think?

agramfort, 2022-05-05:
A plot_ch_adjacency function in mne.viz would be useful yes 👍

alexrockhill, 2022-05-05:
Definitely!

hoechenberger, 2022-05-05:
Yes sounds good! Although for many sensors, I'm not sure how useful it will be?

sappelhoff, 2022-05-06:
> Although for many sensors, I'm not sure how useful it will be?

True, for > 64 sensors I expect this to become more and more crowded ... perhaps useless starting from 100. We'll have to see :eye:

sappelhoff, 2022-05-08:
cc @mmagnuski --> you seem to already have something like this in your "sarna" repo :-) --> https://github.com/mmagnuski/sarna/blob/2bdc6d9da8c8b72b4baf0473ace1ddb89e55e2aa/sarna/cluster.py#L48

mmagnuski, 2022-05-08:
@sappelhoff Haha, yes, I do. :) I was thinking of moving it over to mne long time ago, I can try next (upcoming) week.

PR #10633 (2022-05-22): ENH: plot adjacency
#### Reference issue
Fixes #10602.

#### What does this implement/fix?
This adds `plot_ch_adjacency`. Currently this is a clean-up, fixed version of what I had in my `sarna` repo. In matplotlib interactive mode it allows to edit the adjacency too:
[image]
Currently a bit more work is needed, see below:

#### TODOs:
- [x] rename to `plot_ch_adjacency`
- [x] 3d mode (`kind='3d'`) should turn off edit mode as poistion of mouse clicks are not reliable in three dimensions in matplotlib
- [x] currently if adjacency is sparse it is turned to array. This makes sense for convenient  [... cut]
closed 2022-05-22 by sappelhoff