mne-tools/mne-python #10608: Viz: `evoked.plot_image` with the `group_by` parameter produces faulty sub-panel titles.

sappelhoff, 2022-05-06:
#### Describe the bug

`evoked.plot_image` with the `group_by` parameter produces faulty sub-panel titles.

#### Steps to reproduce

```Python
import mne

path = mne.datasets.eegbci.load_data(1, 1)[0]

raw = mne.io.read_raw_edf(path)
mne.datasets.eegbci.standardize(raw)
raw.set_montage("standard_1005")

epochs = mne.make_fixed_length_epochs(raw)
evoked = epochs.average()

sel = mne.channels.make_1020_channel_selections(evoked.info)

fig, axs = plt.subplots(3, 1)
axes = {sel: ax for sel, ax in zip(sel, axs.ravel())}
evoked.plot_image(group_by=sel, axes=axes, show=False)
fig.tight_layout()
```

#### Expected results

A plot with 3 sub-panels. the titles should be "Left", "Midline", "Right" ... pretty much as here: https://mne.tools/stable/auto_examples/visualization/roi_erpimage_by_rt.html#sphx-glr-auto-examples-visualization-roi-erpimage-by-rt-py

#### Actual results

All sub-panels have the title "Left"

[image]

This is also the case here: https://mne.tools/stable/auto_tutorials/stats-sensor-space/20_erp_stats.html

#### Additional information

happens on mne CI in `main` branch (2022-05-06)

sappelhoff, 2022-05-06:
I believe the error resides somewhere here: https://github.com/mne-tools/mne-python/blob/add9e1620958b61f5509bea172ce8343975aee10/mne/viz/evoked.py#L238-L239

hoechenberger, 2022-05-07:
Cool, can you triple-check it's really just the labels that are wrong, or if there's maybe even an issue with the displayed data?

hoechenberger, 2022-05-07:
Also for EEG, I don't believe a "channel index" on the Y axis is helping anyone looking at this plot…

sappelhoff, 2022-05-07:
Judging from my plots, the data is displayed correctly.

> Also for EEG, I don't believe a "channel index" on the Y axis is helping anyone looking at this plot…

you can pass `show_names="all"` to fix this (see [API](https://mne.tools/stable/generated/mne.Evoked.html#mne.Evoked.plot_image)), but then you have to make the axes bigger as well (or tick label font smaller), else the channel labels overlap by too much

closed 2022-05-12 by larsoner