mne-tools/mne-python #10612: Use spatial_colors by default in Evoked.plot()

hoechenberger, 2022-05-07:
When using Evoked.plot(), in fact I **always** want to use `spatial_colors=True` as this feature is so simple, yet extremely helpful.

Problem is, on the first try I almost always **forget** to pass the respective parameter, so then I need to amend my code and re-run it.

Also I believe that many users are probably not aware of the feature, as … well, as it's not the default.

Could we not make it such that if we have sensor locations, we use `spatial_colors=True` by default?

We'd still keep the parameter to allow for manual adjustment, of course.

WDYT?

cc @sappelhoff @cbrnr @alexrockhill @adam2392

cbrnr, 2022-05-07:
Fine with me!

sappelhoff, 2022-05-07:
Fine with me as well :+1:

alexrockhill, 2022-05-07:
Definite +1

agramfort, 2022-05-07:
No objection but make sure no warning is raised by default if no channel locations are present 

This maybe suggests a spatial_colors=‘auto’ by default

mmagnuski, 2022-05-07:
+1 from me too.

drammock, 2022-09-14:
reopened by #11151

PR #11017 (2022-09-07): [MRG, ENH] Use spatial_colors by default in Evoked.plot()
#### Reference issue
Fixes #10612 with respect to [comment](https://github.com/mne-tools/mne-python/issues/10612#issuecomment-1120288827)

#### What does this implement/fix?
Adds a default 'auto' option for spatial_colors argument in Evoked.plot().
Which makes spatial_colors True if channel locations exists and False otherwise.
Changes docstring in plot_evoked() to include string
Adds option chek to plot_evoked for spatial_colors argument

#### Additional information
Any additional information you think is important.
PR #11201 (2022-09-28): Fix spatial colors
closes #10612
replaces/supersedes #11017
closed 2022-09-28 by larsoner