mne-tools/mne-python #10623: MNE Viz Raw Browser crashes due to NameError: name '_qt_init_icons' is not defined

Taremeh, 2022-05-11:
#### Describe the bug
Freshly installed MNE Version 1.0.2 for MacOS M1 / Apple Silicon (using Anaconda) causes the following error when trying to call `raw.plot()` if the MacOS theme option is set to dark mode:

```sh
Traceback (most recent call last):
  File "/Users/tarek/Documents/Development/BA/eeg-mne/topomap.py", line 27, in <module>
    raw.plot(block=True )
  File "/Users/tarek/opt/anaconda3/envs/mne/lib/python3.10/site-packages/mne/io/base.py", line 1550, in plot
    return plot_raw(self, events, duration, start, n_channels, bgcolor,
  File "<decorator-gen-182>", line 12, in plot_raw
  File "/Users/tarek/opt/anaconda3/envs/mne/lib/python3.10/site-packages/mne/viz/raw.py", line 356, in plot_raw
    fig = _get_browser(show=show, block=block, **params)
  File "/Users/tarek/opt/anaconda3/envs/mne/lib/python3.10/site-packages/mne/viz/_figure.py", line 654, in _get_browser
    fig = backend._init_browser(**kwargs)
  File "/Users/tarek/opt/anaconda3/envs/mne/lib/python3.10/site-packages/mne_qt_browser/_pg_figure.py", line 4748, in _init_browser
    browser = MNEQtBrowser(**kwargs)
  File "/Users/tarek/opt/anaconda3/envs/mne/lib/python3.10/site-packages/mne_qt_browser/_pg_figure.py", line 221, in func
    meth(self, *args, **kwargs)
  File "/Users/tarek/opt/anaconda3/envs/mne/lib/python3.10/site-packages/mne_qt_browser/_pg_figure.py", line 2713, in __init__
    stylesheet = _qt_get_stylesheet(getattr(self.mne, 'theme', 'auto'))
  File "/Users/tarek/opt/anaconda3/envs/mne/lib/python3.10/site-packages/mne/viz/backends/_utils.py", line 235, in _qt_get_stylesheet
    icons_path = _qt_init_icons()
NameError: name '_qt_init_icons' is not defined
```
Due to this error, the raw browser can't open.

#### Steps to reproduce
Independent of the data set used, calling the `raw.plot()` method on MacOS with the OS theme set to `dark` will cause the described error, crashing the raw browser.

#### Expected results
The raw browser should open itself on method call `raw.plot()` as it is working when the MacOS theme is set to `light`.

#### Additional information
This code causes the error (`/mne/viz/backends/_utils.py:235`):
```python
def _qt_get_stylesheet(theme):
    from ...fixes import _compare_version
    from ...utils import logger, warn, _validate_type
    _validate_type(theme, ('path-like',), 'theme')
    theme = str(theme)
    system_theme = None
    if theme == 'auto':
        theme = system_theme = _qt_detect_theme()
    if theme in ('dark', 'light'):
        if system_theme is None:
            system_theme = _qt_detect_theme()
        if sys.platform == 'darwin' and theme == system_theme:
            from qtpy import QtCore
            try:
                qt_version = QtCore.__version__  # PySide
            except AttributeError:
                qt_version = QtCore.QT_VERSION_STR  # PyQt
            if theme == 'dark' and _compare_version(qt_version, '<', '5.13'):
                # Taken using "Digital Color Meter" on macOS 12.2.1 looking at
                # Meld, and also adapting (MIT-licensed)
                # https://github.com/ColinDuquesnoy/QDarkStyleSheet/blob/master/qdarkstyle/dark/style.qss  # noqa: E501
                # Something around rgb(51, 51, 51) worked as the bgcolor here,
                # but it's easy enough just to set it transparent and inherit
                # the bgcolor of the window (which is the same). We also take
                # the separator images from QDarkStyle (MIT).
                icons_path = _qt_init_icons() # <--- NameError: name '_qt_init_icons' is not defined
                stylesheet = """\
...
```

```
mne.sys_info()
Platform:         macOS-12.2-x86_64-i386-64bit
Python:           3.10.4 | packaged by conda-forge | (main, Mar 24 2022, 17:45:10) [Clang 12.0.1 ]
Executable:       /Users/tarek/opt/anaconda3/envs/mne/bin/python
CPU:              i386: 8 cores
Memory:           16.0 GB
mne:              1.0.2
numpy:            1.21.6 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:            1.8.0
matplotlib:       3.5.2 {backend=module://backend_interagg}
sklearn:          1.0.2
numba:            0.55.1
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             1.5.0
cupy:             Not found
pandas:           1.4.2
pyvista:          0.34.0 {OpenGL 4.1 Metal - 76.3 via Apple M1 Pro}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.2
```

#### Quick Fix: How To 😉 
1. Set your MacOS theme to `light` to avoid getting into the erroneous `if theme == 'dark'` clause.
2. Done ;)

larsoner, 2022-05-11:
Argh yes this is caused by a bad backport by me. We could either remove this code (which fixes toolbar colors), or backport the icon repackaging that @GuillaumeFavelier implemented. @hoechenberger which do you think makes the most sense for 1.0.3?

hoechenberger, 2022-05-11:
@larsoner It depends on when we're planning to release MNE 1.1…
If it's another few weeks, I'd say let's try to fix the backport and get 1.0.3 out ASAP, WDYT?

hoechenberger, 2022-05-11:
@Taremeh Super helpful and complete bug report, btw! 👍 I'd love to see many more of those 😁

hoechenberger, 2022-05-13:
This is fixed in 1.0.3

PR #10621 (2022-05-11): MRG: Backport #10619 and #10620
Closes #10623
closed 2022-05-13 by hoechenberger