mne-tools/mne-python #10629: Save dist_to_inner_skull in fit_dipole() in the Dipole instance as attribute

mjcourte, 2022-05-13:
#### Describe the new feature or enhancement

During the mne.fit_dipole() procedure, with verbose=1, it prints the depth from inner skull of each fit. This is information I would like to be able to save and retrieve later for comparisons of depth vs. SNR and the like. As far as I can tell, this isn't saved in any way, only used for the testing against the min_dist constraint.

#### Describe your proposed implementation

When fit_dipole() creates an instance of Dipole, save dist_to_inner_skull as an attribute as it is calculated, just like the other dipole fit parameters (pos, ori, etc.).

#### Describe possible alternatives

- Expose mne.surfaces._compute_nearest() in an accessible wrapper. For example I pass some args (Dipole, bem, etc) and retrieve the depths. As these are already computed during fit_dipole, I think the min work solution is just to assign dist_to_inner_skull as an attribute at time of fit.
- I have been just using linalg find nearest point manually. Turn the bem surf into a pointcloud, subtract dipole pos vector from each point, norm, find min.

Thanks,
Matt

larsoner, 2022-05-13:
Rather than add something like this, I'm inclined to add a new function like:
```
mne.bem.distance_to_bem(pos, bem, trans=None)
```
where `pos` is in head coordinates (which is the case for `dip.pos`), `bem` can be a BEM or sphere model, and `trans` can be provided. Then it computes the distances from the inner skull BEM surface to `pos`.

This should be very fast to compute and ends up being more general because it can be used for any position in head coordinates, not just those from a dipole.

Would you be interested in implementing this @mjcourte ?

larsoner, 2023-09-25:
Done in #10632

closed 2023-09-25 by larsoner