mne-tools/mne-python #10631: Cropping Brainvision dataset results in incorrect annotations

davidhaslacher, 2022-05-13:
Annotations are sometimes not adjusted correctly when cropping a Brainvision dataset. For instance, in the example below, the last event is originally inside the data range. After cropping, the last event is outside of the data range.

https://drive.google.com/file/d/1csDVfPNQB5C-TJJ9jV7JyHPlFnu-mHPe/view?usp=sharing

hoechenberger, 2022-05-14:
Hello, do you still see this with 1.0.3 and `main`? We'd recently fixed an issue related to annotations and cropping

davidhaslacher, 2022-05-15:
Hello! I just tested this, and the issue unfortunately persists.

agramfort, 2022-05-15:
hi @davidhaslacher the behavior you observe is expected. The events are defined taking into account the raw.first_samp attribute. When you crop you just adjust the first_samp the events don't change. The historical motivation is specifically to have events who do not depend on the crop but the absolute time of the recording. 

See an updated script:

```
import mne

path = '4_80.vhdr'

raw = mne.io.read_raw_brainvision(path, preload=True)

n_times_before_crop = raw.n_times
events_stim_before_crop = mne.events_from_annotations(raw)[0]

print(raw.first_samp)
print(n_times_before_crop)
print(events_stim_before_crop[-1][0] - raw.first_samp)

raw.crop(60, 120)
n_times_after_crop = raw.n_times
events_stim_after_crop = mne.events_from_annotations(raw)[0]

print(raw.first_samp)
print(n_times_after_crop)
print(events_stim_after_crop[-1][0] - raw.first_samp)
```

hoechenberger, 2022-05-15:
@agramfort
OP was reporting unexpected behavior with annotations, not events, when cropping. Therefore I thought maybe it's yet another of those bugs we recently tried to address.

davidhaslacher, 2022-05-15:
Hi @agramfort, thanks for clearing that up. @hoechenberger, given that this is the expected behavior, I don't see a problem anymore. However, I find it quite confusing that events after cropping are relative to the start of the original recording, not the start of the cropped recording. Perhaps this could be selected with an appropriate parameter to events_from_annotations.

closed 2022-05-15 by davidhaslacher

hoechenberger, 2022-05-15:
Oh, seems I entirely misunderstood your question here. But glad to hear it's been mostly sorted out!