mne-tools/mne-python #10648: BUG: Memmapping tests fail

larsoner, 2022-05-18:
```bash
pytest mne\io\fiff\tests\test_raw_fiff.py
=========================================================================================================== test session starts ============================================================================================================
platform win32 -- Python 3.10.4, pytest-7.1.1, pluggy-1.0.0
rootdir: C:\Users\matti.toivonen\src\mne-python, configfile: setup.cfg
plugins: cov-3.0.0, harvest-1.10.3, timeout-2.1.0
collected 67 items

mne\io\fiff\tests\test_raw_fiff.py ..............................................s.......FF..FF....s..                                                                                                                                [100%]
```

The failures occur in other test-functions and share the following error message:

```bash
______________________________________________________________________________________________________ test_file_like[True-str-file] _______________________________________________________________________________________________________
mne\io\fiff\tests\test_raw_fiff.py:1726: in test_file_like
    _test_raw_reader(read_raw_fif, **kwargs)
mne\io\tests\test_raw.py:413: in _test_raw_reader
    _test_raw_crop(reader, t_prop, kwargs)
mne\io\tests\test_raw.py:423: in _test_raw_crop
    raw_1 = reader(**kwargs)
mne\io\fiff\raw.py:482: in read_raw_fif
    return Raw(fname=fname, allow_maxshield=allow_maxshield,
<decorator-gen-241>:12: in __init__
    ???
mne\io\fiff\raw.py:125: in __init__
    self._preload_data(preload)
mne\io\base.py:566: in _preload_data
    self._data = self._read_segment(
<decorator-gen-205>:12: in _read_segment
    ???
mne\io\base.py:394: in _read_segment
    data = _allocate_data(data_buffer, data_shape, dtype)
mne\io\base.py:2044: in _allocate_data
    data = np.memmap(str(preload), mode='w+', dtype=dtype, shape=shape)
..\..\Miniconda3\envs\mnedev\lib\site-packages\numpy\core\memmap.py:228: in __new__
    f_ctx = open(os_fspath(filename), ('r' if mode == 'c' else mode)+'b')
E   OSError: [Errno 22] Invalid argument: 'C:\\Users\\matti.toivonen\\AppData\\Local\\Temp\\pytest-of-matti.toivonen\\pytest-4\\test_file_like_True_str_file_0\\memmap'
```

_Originally posted by @mattitoi in https://github.com/mne-tools/mne-python/issues/10639#issuecomment-1129661496_

larsoner, 2022-05-18:
@mattitoi can you post the output of `mne sys_info` here?

Are you running this from a Windows command prompt?

It's weird that you get these failures locally because our CIs do run on Windows. It suggests that there is something to learn here... I don't see anything suspicious in the filename that would cause it to fail to be created in Windows. After it fails, can you try to create the file manually with something like:
```
python -c "open(<samepath>, 'w+b')
```
where `samepath` is whatever path the error message says. I expect this to fail, but then maybe by modifying/trimming/simplifying the path you can figure out if there is some issue along the way. Or maybe try creating a file in some other program like Notepad (or whatever simple editor exists in Windows nowadays) in the same path and see if that succeeds so we can narrow it down to a Python problem (or not).

One thing that just occurred to me -- does that file already exist? Maybe the issue is that we are doing a `np.memmap` on a path that exists, so we need to remove the file first...

mattitoi, 2022-05-19:
@larsoner Yes, running from a Windows command prompt.

Running this command is succesful:

```bash
python -c "open('C:\\Users\\matti.toivonen\\AppData\\Local\\Temp\\pytest-of-matti.toivonen\\pytest-4\\test_file_like_True_str_file_0\\memmap', 'w+b')"
```

The `memmap` files are created upon running of `pytest`. For each run of `pytest` a new folder is created for the run in `C:\Users\matti.toivonen\AppData\Local\Temp\pytest-of-matti.toivonen\pytest-X`, where is a running number.

Output of `mne sys_info`

```bash
(mnedev) C:\Users\matti.toivonen\src\mne-python>mne sys_info
Platform:         Windows-10-10.0.19044-SP0
Python:           3.10.4 | packaged by conda-forge | (main, Mar 24 2022, 17:34:17) [MSC v.1929 64 bit (AMD64)]
Executable:       C:\Users\matti.toivonen\Miniconda3\envs\mnedev\python.exe
CPU:              Intel64 Family 6 Model 165 Stepping 2, GenuineIntel: 12 cores
Memory:           31.8 GB

mne:              1.1.dev0
numpy:            1.21.6 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:            1.8.0
matplotlib:       3.5.2 {backend=QtAgg}

sklearn:          1.1.0
numba:            0.55.1
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             1.5.0
cupy:             Not found
pandas:           1.4.2
pyvista:          0.34.0 {OpenGL 4.5.0 - Build 30.0.100.9864 via Intel(R) UHD Graphics}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.2
vtk:              9.1.0
qtpy:             2.1.0 {PyQt5=5.12.9}
ipympl:           Not found
pyqtgraph:        0.12.4
pooch:            v1.6.0

mne_bids:         0.11.dev0
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.3.1
mne_connectivity: 0.4dev0
mne_icalabel:     Not found
```

larsoner, 2023-10-01:
Closing as I haven't seen this on CIs in a year or so, so hoping it was a transient bug that was fixed upstream 🤞 But let's reopen if someone hits it again

closed 2023-10-01 by larsoner