mne-tools/mne-python #10652: Can't open snirf file

ansleykunnath, 2022-05-19:
#### Describe the bug
"write_raw_snirf" and "read_raw_snirf" are not working with my .snirf files from fNIRS recordings with the Kernel Flow system. I am able to use the following code to plot other .snirf sample files found online, but not the .snirf files produced from my fNIRS system.

#### Steps to reproduce
```
from mne_nirs.io import write_raw_snirf
from mne.io import read_raw_snirf

snirf_intensity = read_raw_snirf('Kernel_Flow_Recording.snirf').load_data()
snirf_intensity.load_data()
write_raw_snirf(snirf_intensity, 'Kernel_Flow_Recording.snirf')
snirf_intensity.plot(duration=2, show_scrollbars=False)
```
The file Kernel_Flow_Recording.snirf can be found here: https://github.com/ansleykunnath/fnirs.git

#### Expected results
I expected a plot to pop up displaying the recordings for each channel over time. (This happens with the online sample .snirf files.)

#### Actual results
Python 3.10.4 (v3.10.4:9d38120e33, Mar 23 2022, 17:29:05) [Clang 13.0.0 (clang-1300.0.29.30)]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.3.0 -- An enhanced Interactive Python. Type '?' for help.
PyDev console: using IPython 8.3.0
Python 3.10.4 (v3.10.4:9d38120e33, Mar 23 2022, 17:29:05) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
runfile('/Users/ansle/PycharmProjects/test1/open_snirf_file.py', wdir='/Users/ansle/PycharmProjects/test1')
Backend MacOSX is interactive backend. Turning interactive mode on.
Loading /Users/ansle/PycharmProjects/test1/Kernel_Flow_Recording.snirf
Reading 0 ... 16  =      0.000 ...     2.242 secs...
Traceback (most recent call last):
  File "/Users/ansle/PycharmProjects/test1/venv/lib/python3.10/site-packages/IPython/core/interactiveshell.py", line 3397, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-b2eef07cb0fb>", line 1, in <cell line: 1>
    runfile('/Users/ansle/PycharmProjects/test1/open_snirf_file.py', wdir='/Users/ansle/PycharmProjects/test1')
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_umd.py", line 198, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/Users/ansle/PycharmProjects/test1/open_snirf_file.py", line 17, in <module>
    write_raw_snirf(snirf_intensity, 'Kernel_Flow_Recording.snirf')
  File "/Users/ansle/PycharmProjects/test1/venv/lib/python3.10/site-packages/mne_nirs/io/snirf/_snirf.py", line 29, in write_raw_snirf
    picks = _picks_to_idx(raw.info, 'fnirs_cw_amplitude', exclude=[])
  File "/Users/ansle/PycharmProjects/test1/venv/lib/python3.10/site-packages/mne/io/pick.py", line 1047, in _picks_to_idx
    picks = _picks_str_to_idx(info, picks, exclude, with_ref_meg,
  File "/Users/ansle/PycharmProjects/test1/venv/lib/python3.10/site-packages/mne/io/pick.py", line 1160, in _picks_str_to_idx
    raise ValueError(
ValueError: picks ('fnirs_cw_amplitude') could not be interpreted as channel names (no channel "['fnirs_cw_amplitude']"), channel types (no type "None"), or a generic type (just "all" or "data")

#### Additional information
Platform:         macOS-12.3.1-arm64-arm-64bit
Python:           3.10.4 (v3.10.4:9d38120e33, Mar 23 2022, 17:29:05) [Clang 13.0.0 (clang-1300.0.29.30)]
Executable:       /Users/ansle/PycharmProjects/test1/venv/bin/python
CPU:              arm: 8 cores
Memory:           Unavailable (requires "psutil" package)
mne:              1.0.3
numpy:            1.22.3 {}
scipy:            1.8.0
matplotlib:       3.5.2 {backend=MacOSX}
sklearn:          1.1.0
numba:            Not found
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             Not found
cupy:             Not found
pandas:           1.4.2
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
PyQt5:            Not found
ipympl:           Not found
pooch:            v1.6.0
mne_bids:         Not found
mne_nirs:         0.2.1
mne_features:     Not found
mne_qt_browser:   0.3.1
mne_connectivity: Not found

larsoner, 2022-05-19:
Cc @rob-luke , I'll try to look in the coming days

rob-luke, 2022-05-20:
Your data is in "hbo"/"hbr" type. According to the `write_raw_snirf` documentation "Data to write to file. Must contain only fnirs_cw_amplitude type."

So the `write_raw_snirf`requires data in `fnirs_cw_amplitude` type. It does not support writing other types yet. Would you be willing to open a PR to add support for exporting hbo/hbr? 

We should also make the error more informative than it is now

ansleykunnath, 2022-05-20:
Created a pull request. Thank you for the information!