mne-tools/mne-python #10657: Inconsistency in ICA matrix shapes between ICA loaded from EEGLAB and fitted in MNE

mscheltienne, 2022-05-20:
I noticed this issue today, you can reproduce it below with a 32 channels raw and a 25 components ICA.

```
from mne.datasets import sample, testing
from mne.io import read_raw_eeglab, read_raw_fif
from mne.preprocessing import read_ica_eeglab, ICA

test_base_dir = testing.data_path(download=False)

# Load raw and ICA from EEGLAB
fname = testing.data_path(download=False) / "EEGLAB" / "test_raw.set"
raw1 = read_raw_eeglab(fname, preload=True)
ica1 = read_ica_eeglab(fname)
n_ch = len(raw1.ch_names)  # 32
n_components = ica1.n_components  # 25

# Fit an ICA with same n_ch and n_components
fname = sample.data_path(download=False) / 'MEG' / 'sample' / 'sample_audvis_raw.fif'
raw2 = read_raw_fif(fname)
raw2.pick_types(eeg=True)
raw2.pick_channels(raw2.ch_names[:n_ch])
raw2.load_data()
raw2.filter(1., 40.)
ica2 = ICA(n_components=n_components, method='picard')
ica2.fit(raw2)

# Compare
assert ica1.unmixing_matrix_.shape == ica2.unmixing_matrix_.shape
assert ica1.pre_whitener_.shape == ica2.pre_whitener_.shape
assert ica1.pca_mean_.shape == ica2.pca_mean_.shape

# fails:
assert ica1.pca_components_.shape == ica2.pca_components_.shape
# ica1: (25, 32) --- ica2: (32, 32)
assert ica1.pca_explained_variance_.shape == ica2.pca_explained_variance_.shape
# ica1: (25,)    --- ica2: (32,)
```

Probably a problem here: https://github.com/mne-tools/mne-python/blob/965521831b5ca8675becc0cc9716db0cfc8ad946/mne/preprocessing/ica.py#L2830-L2848

agramfort, 2022-05-21:
I don't have time to look in details but MNE does something specific:
it allows to run ICA after PCA but to put back the small PCA components
back into the data after rejection some components.

The example is this: EOG components are super easy and fast to pick up with
20 ICA
components but I don't want to work with data in rank 20 after cleaning the
EOG.

makes sense? does it help?

Message ID: ***@***.***>
>

mscheltienne, 2022-05-21:
Yes, I'm familiar with this approach and it makes perfect sense. Are you saying that this is not the approach chosen by EEGLAB, and thus the small PCA components are missing from the loaded instance?
That would explain the shapes observed in the example above.

agramfort, 2022-05-21:
yes exactly. EEGLAB does not allow to put back the extra PCA components
that's why
It's smaller

> Message ID: ***@***.***>
>

mscheltienne, 2022-05-21:
OK, thanks a lot, so the shape mismatch is normal!

closed 2022-05-21 by mscheltienne