mne-tools/mne-python #10659: BUG: Brain from pyvistaqt breaks when opened from running QApplication instance on MacOs

marsipu, 2022-05-20:
#### Describe the bug
When an instance of `mne.viz.Brain` is shown from within an already running instance of `QApplication`, the python interpreter crashes with a SegmentationError. This happens only on MacOs.

#### Steps to reproduce
```python
import sys

import mne
from PyQt5.QtWidgets import QMainWindow, QWidget, QVBoxLayout, QPushButton, \
    QApplication

mne.datasets.fetch_fsaverage()

def plot_brain():
    Brain = mne.viz.get_brain_class()
    brain = Brain('fsaverage')

class TestWindow(QMainWindow):
    def __init__(self):
        super().__init__()

        widget = QWidget()
        layout = QVBoxLayout(widget)
        test_bt = QPushButton('Plot Brain')
        test_bt.clicked.connect(plot_brain)
        layout.addWidget(test_bt)
        self.setCentralWidget(widget)

def main():
    app = QApplication.instance()
    if app is None:
        app = QApplication(sys.argv)
    window = TestWindow()
    window.show()
    app.exec()

if __name__ == '__main__':
    main()

```

#### Expected results
Brain is shown and the already running QApplication works with `pyvistaqt`.

#### Actual results
Somehow interacting with the opened Brain-Window (moving the mouse) causes a `SegmentationError` and breaks the python interpreter.

<details>
<summary>SegmentationError</summary>
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               python3.10 [1841]
Path:                  /Users/USER/*/python
Identifier:            python3.10
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        zsh [1815]
Responsible:           Terminal [1812]
User ID:               501

Date/Time:             2022-05-20 20:44:28.4389 +0200
OS Version:            macOS 12.3.1 (21E258)
Report Version:        12
Anonymous UUID:        BF4802CD-9CFE-42B1-859F-25CDBA0EE5BD

Time Awake Since Boot: 1300 seconds

System Integrity Protection: enabled
[... 427 lines cut ...]
numpy:            1.21.5 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:            1.8.0
matplotlib:       3.5.1 {backend=MacOSX}

sklearn:          1.0.2
numba:            0.55.1
nibabel:          3.2.2
nilearn:          0.9.0
dipy:             1.5.0
cupy:             Not found
pandas:           1.4.2
pyvista:          0.33.3 {OpenGL 4.1 INTEL-18.5.8 via Intel(R) Iris(TM) Plus Graphics 655}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.2
vtk:              9.1.0
qtpy:             2.0.1 {PyQt5=5.12.3}
ipympl:           Not found
pooch:            v1.6.0

mne_bids:         0.11.dev0
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.2.7.dev0
mne_connectivity: 0.4dev0
</details>

marsipu, 2022-05-20:
@GuillaumeFavelier Do you know something about this problem?

marsipu, 2022-05-20:
With `mne==0.24.1` this bug doesn't seem to appear

hoechenberger, 2022-05-23:
It only crashed when the window received an event, it seems (in my case: moving the mouse cursor), like you said. This is on macOS (btw, this is the way it's supposed to be spelled ;)) 12.3.1

https://user-images.githubusercontent.com/2046265/169799833-48fc076c-6ea3-4109-a5d2-47b3b0cafd0c.mov

marsipu, 2022-05-23:
@larsoner Thank you, I tried to get behind it but this segmentation fault is so hard to debug since it gives little clue what the problem is

larsoner, 2022-05-24:
I added a `import faulthandler; faulthandler.enable()` but it didn't provide any additional information. I also added some debugging prints at various points in `_brain`, `pyvista`, and `pyvistaqt` and couldn't figure this one out. I wonder if we need a debug build of Qt where we can get a useful traceback :(

larsoner, 2022-05-24:
Okay I say that.... then almost immediately look up debugging on macOS and get a bit farther:
```
$ lldb python
(lldb) process launch -- -u /Users/larsoner/Desktop/rep.py
...
2022-05-24 16:15:37.299227-0400 python[91919:796914] [qt.qpa.drawing] Layer-backing is always enabled.  QT_MAC_WANTS_LAYER/_q_mac_wantsLayer has no effect.
Process 91919 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x843e643d642455)
    frame #0: 0x0000000154953088 libvtkRenderingOpenGL2-9.1.9.1.0.dylib`-[vtkCocoaGLView modifyDPIForBackingScaleFactorOfWindow:] + 164
libvtkRenderingOpenGL2-9.1.9.1.0.dylib`-[vtkCocoaGLView modifyDPIForBackingScaleFactorOfWindow:]:
->  0x154953088 <+164>: ldr    x2, [x8, #0x1f0]
    0x15495308c <+168>: ldp    x29, x30, [sp, #0x20]
    0x154953090 <+172>: ldp    x20, x19, [sp, #0x10]
    0x154953094 <+176>: ldp    d9, d8, [sp], #0x30
```
Next up is probably to look into why the DPI modification here causes a segfault.

hoechenberger, 2022-05-24:
So it's likely a VTK bug?

larsoner, 2022-05-24:
Yeah, let's see what they say

https://gitlab.kitware.com/vtk/vtk/-/issues/18559

larsoner, 2023-10-01:
The related VTK issue was closed a year ago and I just tried the MWE at the top and now it's fine!

closed 2023-10-01 by larsoner