mne-tools/mne-python #10665: mne.preprocessing.find_bad_channels_maxwell() should have an n_jobs parameter

hoechenberger, 2022-05-23:
If the user passes a `hfreq`, `find_bad_channels_maxwell()` performs frequency filtering using `Raw.filter()`. This method defaults to `n_jobs=None`, and can therefor happen to start consuming large amounts of memory. We're currently dealing with this in https://github.com/mne-tools/mne-bids-pipeline/pull/557, where my workaround is to wrap `find_bad_channels_maxwell()` in a `joblib.parallel_backend` context (CI is still running, so I'm not 100% sure this will fix the problem)

I propose to add an `n_jobs=None` parameter to `find_bad_channels_maxwell()` that will be passed to `Raw.filter()` to make this easier.

hoechenberger, 2022-05-23:
Update: using the context manager did NOT solve the issue; workaround was to manually filter with n_jobs=1 and then passing the filtered data and hfreq=None to this function. 

So this could also mean that there's a bug here somewhere in MNE's prallelism for filtering. 

cc @larsoner

larsoner, 2022-05-23:
> Update: using the context manager did NOT solve the issue; workaround was to manually filter with n_jobs=1 and then passing the filtered data and hfreq=None to this function.

The first thing to check is how many preloaded instances you have in both cases.

If you had before:
```
raw = read_raw_fif(..., preload=True)
find_bads_maxwell(raw, h_freq=..., n_jobs=1)
```
you would end up with *two* copies of preloaded data in memory because internally `find_bads_maxwell` would make a copy internally to do its filtering and find bads. Have you checked to see if this is the case? You should do this before looking deeper into the filtering parallel logistics.

If you aren't doing this, then it would help to take it down to a minimal example that shows the problem (probably just 2-3 lines to load and then call `find_bads_maxwell`).

hoechenberger, 2022-05-23:
Instance was preloaded in both scenarios. I will see if I can put together a MWE tomorrow

larsoner, 2022-05-23:
To me it sounds like there is no bug

larsoner, 2022-05-23:
(so no need for a MWE, just adjust your usage so that only a single copy of the data is loaded in memory rather than two)

hoechenberger, 2022-05-24:
Thanks @larsoner, this makes sense.

closed 2022-05-24 by hoechenberger