mne-tools/mne-python #10684: [ENH, BUG] Adding to and Fixing Blender Tutorial

alexrockhill, 2022-05-27:
First, the link to the Blender YouTube tutorials has broken here https://mne.tools/dev/auto_tutorials/forward/80_fix_bem_in_blender.html and also in general Blender has updated to version 3.0 which as I've heard is a top to bottom rewrite so I think everything in a screenshot is out of date or at least should be checked that things are still in the same place.

Second, and relatedly on a larger note, I think even though `pyvista` and `mne.viz.Brain` are amazing, it would be nice to have a more general example to leverage the super figure generating power of Blender. I think MEG, EEG and iEEG (and maybe also fNIRS) would be super cool to export as `.obj` files with more of the style of once you do this, go at it however you like in Blender. I think this more hands off approach and treating Blender as just the plotting software for the last step is a lot more maintainable since Blender is pretty fast-moving than these tutorial steps with screenshots about how to edit surfaces. I do think they are super useful but I think the overhead with checking them every Blender version and updating screenshots/remembering to check that they work since it's not covered by CIs is a bit cumbersome. I think it might be a conversation worth having to do a bit of refactoring to make sure things don't get out of date, but still contain very useful information in the tutorial while adding an example with sensors and not just surfaces.

wmvanvliet, 2022-06-06:
I am in favor of cutting it down. But let's have some links to specific parts of the Blender documentation that are helpful. Blender is a huge software package.

hoechenberger, 2022-06-07:
I've added this to the 1.1 milestone, is anyone actively working on this?

wmvanvliet, 2022-06-07:
> Yes! It might be more portable to put the color of the vertices in an obj file though so that there isn't any bpy scripting if it can be avoided.

Is there a way to animate that?

wmvanvliet, 2022-06-07:
@hoechenberger I think we can remove the specific blender instructions and instead point to specific parts of the official blender documentation quite quickly. Building a better MNE-Python <-> Blender bridge (either by exporting files that Blender can read, or scripting Blender to read MNE-Python object) will take more time and should probably not be scheduled for the next release.

alexrockhill, 2022-06-07:
> > Yes! It might be more portable to put the color of the vertices in an obj file though so that there isn't any bpy scripting if it can be avoided.
> 
> Is there a way to animate that?

I'd really want to look into that, that would be very cool and portable!

alexrockhill, 2022-06-07:
> @hoechenberger I think we can remove the specific blender instructions and instead point to specific parts of the official blender documentation quite quickly. Building a better MNE-Python <-> Blender bridge (either by exporting files that Blender can read, or scripting Blender to read MNE-Python objects) will take more time and should probably not be scheduled for the next release.

I would like to see fixing the links done before the release. +1

alexrockhill, 2022-06-10:
Sorry I've been busier than I thought, maybe just push this to 1.2.

hoechenberger, 2022-06-11:
No worries, @alexrockhill! Will push

closed 2022-06-15 by hoechenberger