mne-tools/mne-python #10701: Error: "Desired compensation matrix not found" - Source Localization with ref_meg=False

RobertoFelipeSG, 2022-06-02:
Hello

My issue is likely related to #5221.

I would like to perform analyses in the source space from a CTF dataset, however I am not able to create a forward solution for any of my subjects given the fact that during my pre processing steps, I chose not to include the reference channels. This has resulted in the error: "Desired compensation matrix not found" when running the make_forward_solution command with a transformation matrix that equals an identity matrix (i.e. =None), or in an error ponting out that there might be "1 MEG sensor inside the scalp surface" when using ='fsaverage' from your datasets. For the records, I don't have right now the MRI scans of my subjects, therefore I am solely relying on the head models 'fsaverage' and 'spm' that you guys provide.

Is there a way to work around this? For instance, calculating and adding such a compensation matrix to the epoch files before going to the source space? I would like to avoid, if possible re running from scratch the preprocessing. 

Thanks in advance.

Rob

agramfort, 2022-06-04:
@RobertoFelipeSG what was the reason for dropping the ref_meg channels? how did you end up removing the compensation matrix? before trying to hack it back in let's see if it's easier to never have lost it.

my 2c

RobertoFelipeSG, 2022-06-06:
@agramfort Thanks for your willingness to help. I didn't include the ref_meg because our recordings include a 3rd level gradient compensation. I did not deliberately remove the compensation matrix. I don't really know if the matrix was dropped during any of my preprocessing steps following a standard pipeline. Would that have been possible through any of the MNE commands (e.g. Filtering, Finding events, Making a montage, Picks)?

larsoner, 2022-06-07:
> I didn't include the ref_meg because our recordings include a 3rd level gradient compensation. I did not deliberately remove the compensation matrix.

When you pick channels without including the reference channels, their associated compensation matrices will be removed, too. If you want to use gradient-compensated data, in general you should keep the reference channels. Is it possible to change your pipeline to do this?

RobertoFelipeSG, 2022-06-08:
Dear @larsoner, thanks for the explanation. I re ran my preprocessing pipeline in one of my datasets including the ref_meg (i.e =True) channels, hoping that this would solve the issue later on during the source reconstruction. However, I always get the same error when creating the forward solution: "Found N sensors inside the scalp surface, perhaps coordinate frames and/or coregistration must be incorrect". I tried with the SPM and fsaverage datasets and the only difference is the number of channels N that are found outside the scalp surface when the error message pops up. Any ideas on how to make this work?

From one side, I would prefer not to re run all the preprocessing in all datasets for the sake of time. Nonetheless, if this would solve the issue, I am willing to start from scratch. From the other side, now I am doubting that the ref_meg channels are really the missing link that would permit me the creation of the forward solution. From a general perspective, what would be your advice to do source reconstruction on CTF data if I don't have access to the MRI of subejcts?

larsoner, 2022-06-08:
> "Found N sensors inside the scalp surface, perhaps coordinate frames and/or coregistration must be incorrect".

This suggests that your CTF system is recording reference sensor positions that are inside the head. Can you check what `plot_alignment` looks like with these data with `meg=('sensors', 'ref')`?

RobertoFelipeSG, 2022-06-13:
@larsoner I am sorry for the delay in my response. To correctly perform the "plot_alignment", I had to deal with some visualization issues in the institutional server I use. In any case, here is the figure resulting from such an alignment. It indeed looks that my MEG recordings and the average MRI head are in two different planes of reference, even though I had included for this example the ref_meg = True in the picks function. What would you suggest to do in this case to align them? How could I possibly define a correct transformation matrix that permits me to perform a source reconstruction ?
Thanks in advance.

[image]

larsoner, 2022-06-14:
> What would you suggest to do in this case to align them?

Either your MRI<->head transform is wrong, which you can fix using `mne coreg`. Or your head<->MEG transform is wrong, which indicates a problem with your recording.

This doesn't seem like an MNE-Python issue at this point, though, so let's continue the conversation over on https://mne.discourse.group/ since it's now about usage I think!

closed 2022-06-14 by larsoner